MarlinFirmware/Marlin #17007: [BUG] M206 values ignored in 2.0.4.3 and 2.0.4.4

codiac2600, 2020-02-27:
### Bug Description

I've set my offsets like I always do and in 2.0.3 they work but in the latest builds they are ignored. 

#define X_MIN_POS -5
#define Y_MIN_POS -12
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 210

Confirmed here:

15:54:50.962: echo:; Home offset:
15:54:50.962: echo: M206 X-5.00 Y-12.00 Z0.00

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Set offsets in firmware
2. ran M502 
3. Ran M500
4. G28

**Expected behavior:** Changing the values would change my home offset

**Actual behavior:** I've tried to -50, -50 and it still stays at the same place.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

zonalimitatore, 2020-02-27:
With my atmega 2560 only on Y axe
00:30:05.628 : X:-9.00 **Y:-6.00** Z:0.00 E:0.00 

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -9
#define Y_MIN_POS **-8**
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 260

thinkyhead, 2020-02-28:
`M206` is not related to the min xy position values. The `M206` offsets are meant to be 0,0 unless you have special need to add an offset to your home position later at runtime.

codiac2600, 2020-02-28:
> `M206` is not related to the min xy position values. The `M206` offsets are meant to be 0,0 unless you have special need to add an offset to your home position later at runtime.

I understand. But my 0,0 isn’t correct and I’m trying to adjust it. When I had it set in 2.0.3 with the values above it worked. In 2.0.4.4 my print head is beyond the bed when it starts to print. Safe homing also isn’t centered.

thinkyhead, 2020-02-28:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-02-28 by thinkyhead

codiac2600, 2020-02-28:
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

Okay so min offset doesn’t work and you’re saying it’s not a bug. I can replicate this every single time. Same firmware in 2.0.3 it works correctly. In 2.0.4.3 and 2.0.4.4 it doesn’t. 

Guess I’ll wait till you eventually fix this when you realize it’s an issue. Thanks?

thinkyhead, 2020-02-28:
You'll need to get help from someone other than this issue queue first and see if you can resolve it with guidance and assistance. We simply can't do that here, and I'm not convinced it's a bug. You'll need to do more testing to prove your case. I sure as heck don't have time to chase it down at the moment.

Vertabreak, 2020-03-14:
M206 is part of the cnc system and does not account for probe offsets the xy min pos generally handles what your trying to do but it would be handled a number of other ways. on the other hand if you where to manually add your probe offsets to the position you where trying to set with m206 it might be a kind of workaround but this isn't the intend.