MarlinFirmware/Marlin #17008: [BUG] (MKS_12864OLED compilation error)

htaxil, 2020-02-27:
### Bug Description

I get this compilation error :

> Compiling .pio\build\LPC1768\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\basic\spi.cpp.o
> 
> Compiling .pio\build\LPC1768\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\command_processor.cpp.o
> In file included from Marlin\src\lcd\dogm\../../inc/../HAL/./HAL_LPC1768/HAL.h:42,
>                  from Marlin\src\lcd\dogm\../../inc/../HAL/HAL.h:26,
>                  from Marlin\src\lcd\dogm\../../inc/MarlinConfig.h:30,
>                  from Marlin\src\lcd\dogm\HAL_LCD_class_defines.h:24,
>                  from Marlin\src\lcd\dogm\ultralcd_DOGM.h:31,
>                  from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:42:
> Marlin\src\lcd\dogm\ultralcd_DOGM.cpp: In static member function 'static void MarlinUI::init_lcd()':
> Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:260:18: error: 'LCD_PINS_DC' was not declared in this scope; did you mean 'LCD_PINS_D6'?
>   260 |       SET_OUTPUT(LCD_PINS_DC);
>       |                  ^~~~~~~~~~~
> Marlin\src\lcd\dogm\../../inc/../HAL/./HAL_LPC1768/fastio.h:52:49: note: in definition of macro 'WRITE_PIN'
>    52 | #define WRITE_PIN(IO,V)       LPC176x::gpio_set(IO, V)
>       |                                                 ^~
> Marlin\src\lcd\dogm\../../inc/../HAL/./HAL_LPC1768/fastio.h:105:35: note: in expansion of macro '_WRITE'
>   105 | #define SET_OUTPUT(IO)        do{ _WRITE(IO, _READ(IO)); _SET_OUTPUT(IO); }while(0)
>       |                                   ^~~~~~
> Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:260:7: note: in expansion of macro 'SET_OUTPUT'
>   260 |       SET_OUTPUT(LCD_PINS_DC);
>       |       ^~~~~~~~~~
> Compiling .pio\build\LPC1768\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\dl_cache.cpp.o
> *** [.pio\build\LPC1768\src\src\lcd\dogm\ultralcd_DOGM.cpp.o] Error 1

### My Configurations
Marlin 2.0.4.4
Mainboard : Re-ARM Controller for RAMPS
RAMPS 1.5
Display : MKS_12864OLED

[MyMarlinConfiguration.zip](https://github.com/MarlinFirmware/Marlin/files/4263953/MyMarlinConfiguration.zip)

### Steps to Reproduce

1. Just launch the compilation

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information
[image]

ellensp, 2020-02-28:
Looks like no one has tried this combination of controller and LCD before. 

I don't have this LCD either, but can make and educated guess.

Try this new Marlin/src/pins/lpc1768/pins_RAMPS_RE_ARM.h
[pins_RAMPS_RE_ARM.zip]
(https://github.com/MarlinFirmware/Marlin/files/4265054/pins_RAMPS_RE_ARM.zip)

borland1, 2020-02-28:
The pins file for that board does not currently support that display.  Take a look at the pins file for board MKS_SGEN_L (also LPC1768 processor).  You might also want to try changing display to MKS_12864OLED_SSD1306 instead in your configuration.h file (should be compatible with SH1106 display controller), as the board pins files are not setup for that display controller.

htaxil, 2020-02-29:
> Try this new Marlin/src/pins/lpc1768/pins_RAMPS_RE_ARM.h
> (https://github.com/MarlinFirmware/Marlin/files/4265054/pins_RAMPS_RE_ARM.zip)

Thanks @ellensp , now the compilation is processing well. But unfortunately, the display remains off...

htaxil, 2020-02-29:
> 
> 
> try changing display to MKS_12864OLED_SSD1306 instead in your configuration.h file

Thanks @borland1 , compilation fails, whether I choose MKS_12864OLED or MKS_12864OLED_SSD1306. Compilation proceeds correctly in both cases if put the pins_RAMPS_RE_ARM.h file provided by @ellensp.

But unfortunately, despite that, the display remains hopelessly black for the moment.

borland1, 2020-02-29:
Your display module uses SPI pinouts at ESP1 and ESP2, so you need to check your controller board's circuit diagram to match up the display pinouts with the controller board's SPI pins at ESP1 and ESP2.  You will need to get that from the manufacturer's circuit diagram which should be posted somewhere on the internet.

Once you have that configured correctly, your display should work.   You can then try both MKS_12864 and MKS_12864OLED_SSD1306 in configuration.h file to see which one works best.

ellensp, 2020-03-01:
Well since I can find no documentation for the LCD, but borland1 says it works on a MKS_SGEN_L I created this conversion document
[lcd_pinout_info.txt](https://github.com/MarlinFirmware/Marlin/files/4271281/lcd_pinout_info.txt)
From this is made some adjustment to the re-arm pins file.
[pins_RAMPS_RE_ARM.h.txt](https://github.com/MarlinFirmware/Marlin/files/4271296/pins_RAMPS_RE_ARM.h.txt)
NB you must enable #define SDSUPPORT in Configuration.h and set #define SDCARD_CONNECTION LCD in Configuration_adv.h to force it to use the SPI bus on the LCD

So give this new pins_RAMPS_RE_ARM.h file a go.  (make sure to remove the .txt file extention) 
NB I am presuming your using a RAMP LCD converter on your Re-ARM.

borland1, 2020-03-01:
on mks_sgen_l
/**
 *                EXP1       _____
 * BEEPER_PIN  1.31 | · ·   | 1.30 BTN_ENC 
 * DOGLCD_MOSI 0.18 | · · | 0.16 DOGLCD_CS 
 * DOGLCD_SCK  0.15 | · · | 0.17 DOGLCD_A0 
 * DOGLCD_RS    1.0 | · · | 1.22 
 *              GND | · · | 5V 
 *                   -----     
 
 *                 EXP2        _____
 * MISO_PIN (SD-card) 0.8 | · · | 0.7  SCK_PIN (SD-card)
 *         BTN_EN1   3.25 | · · | 0.28 SS_PIN(SD_card)
 *         BTN_EN2   3.26 | · · | 1.20 MOSI_PIN (SD-card}
 *     SD_DETECT_PIN 0.27 | · · | RST
 *                    GND | · · | NC
 *                         -----

htaxil, 2020-03-01:
Hi all, @ellensp I have used your pins_RAMPS_RE_ARM.h, #define SDSUPPORT is enable in Configuration.h,  #define SDCARD_CONNECTION LCD is set in Configuration_adv.h. 
for sure i use a RAMP LCD converter

But the display remains turned off. Things happening when I use the button, cause I hear a beep when I push on it, and I get errors on repetier-host when I turn it (Error:!! KILL caused by KILL button/pin). I also get this error : echo : SD init fail.

[image]

This is my display : https://reprap.org/wiki/MKS_12864OLED

I have tried to configure and wire a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, everything works well.

[image]

[image]

[image]

borland1, 2020-03-02:
Your Full Graphic Smart Controller display uses parallel data lines.  The data lines for the MKS_12864OLED are serial (SPI).  **ellensp** got most of the pin mapping correct, but missed the A0 pin.
My previous post shows the EXP1 and EXP2 pin mapping for this SPI display used with MKS_SGEN_L board.  That should help with completing the pin mapping for your RE_ARM board.

ellensp, 2020-03-02:
That pin on RE-ARM is NC (not connected) As noted in https://github.com/MarlinFirmware/Marlin/files/4271281/lcd_pinout_info.txt and  
https://reprap.org/wiki/File:Re-ARM_LCD_adapter.jpg 

I questioned in my doc if it was used or not... 
If it is, you need a custom cable.

borland1, 2020-03-02:
The A0 pin corresponds to D/C pin on the display module.  When the display component of the display module is operated in SPI mode, the A0 pin must be pulled low to ground in software.  That could also be done in hardware by using a jumper wire to pull A0 pin to ground. Then map the A0 pin in firmware to some other unused pin.

boelle, 2020-06-20:
@htaxil is the issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle