MarlinFirmware/Marlin #17009: BTT GTR V1.0 Cant compile

Ski3d, 2020-02-28:
### Bug Description

Cant compile GTR V1.0 Using Bugfix or newest release

### My Configurations

Have only Config the settings below so is basically vanilla marlin to see if i can even get it to compile

/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 3

/**
 * This setting determines the communication speed of the printer.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 115200

// Enable the Bluetooth serial interface on AT90USB devices
//#define BLUETOOTH

// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_GTR_V1_0
#endif

### Steps to Reproduce

1. Set setting above 
2. Build using Visual studio code/PlatformIO using the Env:BIGTREE_GTR_V1_0 in project tasks

**Expected behavior:** Compile 

**Actual behavior:** Failed
 

#### Additional Information
And sorry if its just a dumb mistake but i can compile an SKR 1.3, SKR mini v1,0 and stock ender boards with no problems.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

ellensp, 2020-02-28:
Using my best guess at your config (you should ALWAYS include configuration files, then we could just CTRL-A, CTRL-C, CTRL-V and be guaranteed to have your configuration)
This complies fine. Lots of ugly warning, but it does compile.
It looks like it is failing on the final link. 
This is is probably the windows command line limit. 
Delete the directory Marlin/src/lcd/extensible_ui  (as you not using a display) and try again. if it now compiles... Its the above issue.

ellensp, 2020-02-28:
And for future reference please post text logs!, how can we search for a error in a bunch of images?

Ski3d, 2020-02-28:
Well it did not work like you said, but i found out that if i removed the arrow from this line it compiles just fine..
[image]

Ski3d, 2020-02-28:
Can this be added to the new bug fix? i am not the only one have this same problem and would probably help others.

closed 2020-02-29 by Ski3d

thinkyhead, 2020-02-29:
Currently the line is this:
```ini
platform_packages = framework-arduinoststm32@>=3.107,<4
```

Ski3d, 2020-02-29:
By removing the > in that line, it will compile with no fails.

Ski3d, 2020-02-29:
So it has changed since my last download of the bugfix. So we shouldn't see the same problem as i have?.  When I posted2 days ago ish, I was using the latest bugfix

NAPCAL, 2020-04-11:
Too deep in the folders. Move the main Marlin into the root. The compiler has issues with long paths ends up being truncated but platformio reports back the full path.

Here is my complies for the GTR & Pro, no issues and all are in H:\ 
[image]

NAPCAL, 2020-04-11:
Also, the Bugfix is updated nightly so issues might show up that didn't yesterday.

b3nnyb3n, 2020-05-25:
[image]
Hi guys Im having issues too. Would love to know how youve gotten so far when.... I have compared Official  Marlin / BTT marlin / & BTT gtr Marlin builds. They all point to the Pins file for SKR PRO. Pins totally miss match.

I uploaded a build to my gtr sd card and usb arent fuctioning now. (also fresh outta noob school have alot to learn)