MarlinFirmware/Marlin #17011: Layers not Displaying

JubilantReaper, 2020-02-28:
CR-10 Just updated to 2.0.4.4, Updated from 2.0.3
Used latest configuration files
Modified for BLTouch
Uploaded - First received the EEPROM Error on LCD and corrected using M502, M500. Solved
Now when Printing the Layers are not displaying on the LCD (L=-/-), Since they are not displaying, Octoprint can't locate. 

How can I correct this?

ellensp, 2020-02-28:
As far as I can tell the layer information is just status line updates in the gcode file
Are there any M117 gcode lines in the gcode file produced by your slicer.
If so can you paste a few example lines?

JubilantReaper, 2020-02-28:
Ellensp, Thanks. I'm using Cura and made no changes, but for some reason it is now displaying the layer information. Don't know why anything would have changed with the slicer, but resliced a new file and everything is working as it is supposed it...

closed 2020-02-28 by JubilantReaper