MarlinFirmware/Marlin #17014: Hardware Pin Err

jackshi618, 2020-02-28:
/**
 * Pin mapping for the 1280 and 2560
 *
 *   Hardware Pin  : 02 03 06 07 01 05 15 16 17 18 23 24 25 26 64 63 13 12 46 45 44 43 78 77 76 75 74 73 72 71 60 59 58 57 56 55 54 53 50 70 52 51 42 41 40 39 38 37 36 35 22 21 20 19 97 96 95 94 93 92 91 90 89 88 87 86 85 84 83 82 | 04 08 09 10 11 14 27 28 29 30 31 32 33 34 47 48 49 61 62 65 66 67 68 69 79 80 81 98 99 100
 *   Port          : E0 E1 E4 E5 G5 E3 H3 H4 H5 H6 B4 B5 B6 B7 J1 J0 H1 H0 D3 D2 D1 D0 A0 A1 A2 A3 A4 A5 A6 A7 C7 C6 C5 C4 C3 C2 C1 C0 D7 G2 G1 G0 L7 L6 L5 L4 L3 L2 L1 L0 B3 B2 B1 B0 F0 F1 F2 F3 F4 F5 F6 F7 K0 K1 K2 K3 K4 K5 K6 K7 | E2 E6 E7 xx xx H2 H7 G3 G4 xx xx xx xx xx D4 D5 D6 xx xx J2 J3 J4 J5 J6 J7 xx xx xx xx xx
 *   Logical Pin   : 00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 | 78 79 80 xx xx 84 85 71 70 xx xx xx xx xx 81 82 83 xx xx 72 72 75 76 77 74 xx xx xx xx xx
 */

Hardware Pin  65   Port J2  Logical Pin 72
Hardware Pin  66   Port J3  Logical Pin 72

ellensp, 2020-02-29:
This is just a comment in  Marlin/src/HAL/HAL_AVR/fastio/fastio_1280.h

the code looks correct
```
#define DIO72_PIN   PINJ2
#define DIO72_RPORT PINJ
#define DIO72_WPORT PORTJ
#define DIO72_DDR   DDRJ
#define DIO72_PWM   nullptr

#define DIO73_PIN   PINJ3
#define DIO73_RPORT PINJ
#define DIO73_WPORT PORTJ
#define DIO73_DDR   DDRJ
#define DIO73_PWM   nullptr

```

jackshi618, 2020-02-29:
I know, but this comment is wrong.

ellensp, 2020-02-29:
Yes, but since you gave ZERO information on what you where talking about, I had to check it!

closed 2020-02-29 by jackshi618