MarlinFirmware/Marlin #17015: Encoder Err

jackshi618, 2020-02-28:
MEGA2560 and 12864LCD

#define SERVO0_PIN          72

#define BTN_EN1         31
#define BTN_EN2         33
#define BTN_ENC         35

Encoder does not respond！
But Set
#define SERVO0_PIN          73

#define BTN_EN1         31
#define BTN_EN2         33
#define BTN_ENC         35

Encoder is OK!

ellensp, 2020-03-01:
Not enough information to do anything with.

jackshi618, 2020-03-02:
I designed the board myself.
Set Port 72 connect Servo.
#define SERVO0_PIN 72
LCD REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
        #define LCD_PINS_RS     16
        #define LCD_PINS_ENABLE 17
        #define LCD_PINS_D4     23
        #define LCD_PINS_D5     25
        #define LCD_PINS_D6     27
        #define LCD_PINS_D7     29

        #define BEEPER_PIN      37
        #define BTN_EN1         31
        #define BTN_EN2         33
        #define BTN_ENC         35
        #define SD_DETECT_PIN   49
        #define SDSS            53
        #define KILL_PIN        41
12864 Show OK,but Encoder error!
If define SERVO0_PIN 73 or other
12864 Show OK,but Encoder is OK.
I don't understand.why?

thinkyhead, 2020-03-02:
For issues like this it can help to get a pins report. To do this enable the `PINS_DEBUGGING` option. Then you can use the `M43` command to get a report listing which functions are assigned to all the pins. From this we might get more clue what is happening in your setup.

thinkyhead, 2020-03-02:
One thing that may be affecting your situation is that servo timers are not assigned based on the pin but based on the servo index. So it might be that the timer associated with the servo pin no longer matches up after changing only the pin number. If possible, move your peripheral definition to servo 1 instead of redefining the Servo 0 pin.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle