MarlinFirmware/Marlin #17020: This is all the errors i get when i compile marlin 2.0 with arduino IDE

Magnum8101, 2020-02-29:
Arduino:1.8.12 (Windows 10), Kort:"Anet V1.0"

In file included from C:\Users\Pontus\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/Arduino.h:257:0,

                 from sketch\Marlin.ino.cpp:1:

C:\Users\Pontus\Documents\Arduino\hardware\anet\avr\variants\sanguino/pins_arduino.h:43:0: warning: "TIMER2" redefined

 #define TIMER2  5

 ^

In file included from sketch\Marlin.ino.cpp:1:0:

C:\Users\Pontus\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/Arduino.h:210:0: note: this is the location of the previous definition

 #define TIMER2  6

 ^

In file included from C:\Users\Pontus\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/Arduino.h:257:0,

                 from sketch\Marlin.ino.cpp:1:

C:\Users\Pontus\Documents\Arduino\hardware\anet\avr\variants\sanguino/pins_arduino.h:44:0: warning: "TIMER2A" redefined

 #define TIMER2A 6

 ^

In file included from sketch\Marlin.ino.cpp:1:0:

C:\Users\Pontus\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/Arduino.h:211:0: note: this is the location of the previous definition

 #define TIMER2A 7

 ^

In file included from C:\Users\Pontus\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/Arduino.h:257:0,

                 from sketch\Marlin.ino.cpp:1:

C:\Users\Pontus\Documents\Arduino\hardware\anet\avr\variants\sanguino/pins_arduino.h:45:0: warning: "TIMER2B" redefined

 #define TIMER2B 7

 ^

In file included from sketch\Marlin.ino.cpp:1:0:

C:\Users\Pontus\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/Arduino.h:212:0: note: this is the location of the previous definition

 #define TIMER2B 8

 ^

In file included from C:\Users\Pontus\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/Arduino.h:257:0,

                 from sketch\Marlin.ino.cpp:1:

C:\Users\Pontus\Documents\Arduino\hardware\anet\avr\variants\sanguino/pins_arduino.h:47:0: warning: "TIMER3A" redefined

 #define TIMER3A 8

 ^

In file included from sketch\Marlin.ino.cpp:1:0:

C:\Users\Pontus\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/Arduino.h:214:0: note: this is the location of the previous definition

[... 4928 lines cut ...]
 #define _WRITE(IO,V)          do{ if (&(DIO ## IO ## _RPORT) < (uint8_t*)0x100) _WRITE_NC(IO,V); else _WRITE_C(IO,V); }while(0)

                                                                                 ^

c:\users\pontus\appdata\local\temp\arduino_build_379643\sketch\src\hal\hal_avr\fastio.h:96:31: note: in expansion of macro '_WRITE'

 #define WRITE(IO,V)           _WRITE(IO,V)

                               ^

sketch\src\module/stepper/indirection.h:70:33: note: in expansion of macro 'WRITE'

   #define Y_ENABLE_WRITE(STATE) WRITE(Y_ENABLE_PIN,STATE)

                                 ^

sketch\src\module/stepper/indirection.h:639:33: note: in expansion of macro 'Y_ENABLE_WRITE'

     #define DISABLE_STEPPER_Y() Y_ENABLE_WRITE(!Y_ENABLE_ON)

                                 ^

sketch\src\module/stepper/indirection.h:848:30: note: in expansion of macro 'DISABLE_STEPPER_Y'

 #define DISABLE_AXIS_Y() do{ DISABLE_STEPPER_Y(); DISABLE_STEPPER_Y2(); CBI(axis_know

ellensp, 2020-02-29:
and if you search for error you see " #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'""
So you have selected the wrong board in the arduino IDE

Magnum8101, 2020-02-29:
I have tried to select different boards.
What I did:
Downloaded Marlin 2.0, downloaded configuration for Anet A8 board and putted those *.h in marlin folder. I have selected the Anet V1.0 board under tool > board  (have also tried to compile it with different boards).
I have also tried different Arduino AVR boards versions.
Compile marlin 1.* were no problem.

ellensp, 2020-02-29:
Use the configuration files from https://github.com/MarlinFirmware/Configurations/tree/master/config/examples/Anet/A8

From what little you have posted the MOTHERBOARD setting in Configuration is not set to BOARD_ANET_10 as it should be.

Either that or you copied the files into the wrong place.

closed 2020-02-29 by thinkyhead

lloydw88, 2020-03-06:
does anyone have time to tell me where I went wrong?  I believe this is the message after i tried to compile.  Im a noob if you couldnt tell.  Im pretty good if you give it to me in new dummy terms.  all the vidoes about installing it looked pretty easy.  however, its apparently not.

[printer error.txt](https://github.com/MarlinFirmware/Marlin/files/4297203/printer.error.txt)

lloydw88, 2020-03-06:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4297206/Configuration.zip)

thinkyhead, 2020-03-06:
@lloydw88 — We don't pay attention to comments on closed issues. File a new issue if you found a bug, or use one of these support options:

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y