MarlinFirmware/Marlin #17023: Question on rename of Bigtree SKR 1.3 board names

sbaeder, 2020-02-29:
As part of the below commit, it seems that the board names have changed
While this is OK, it would have been nice to see in the release notes or some other means of a "heads up" for those of us who use the board and update firmware regularly

```
Commits on Feb 4, 2020
STM32duino - Use SDIO for onboard SD (#16756)

@Bob-the-Kuhn
Bob-the-Kuhn committed 25 days ago
```

ellensp, 2020-02-29:
I was also annoyed at how they slipped this in...  Got no response.   Just a few others who where also surprised that they did this..  Join the discord, you can take it up with Bob-the-Kuhn directly!

sjasonsmith, 2020-02-29:
The error message tells you exactly what changed, does it not?

thisismike97, 2020-02-29:
I fully agree they should have mentioned this in the official change log on the release page, however it was not just "slipped" in, there is literally a sanity check that immediately tells you what to change to fix the problem.

> #elif MB(BIGTREE_SKR_V1_3)
>     #error "BOARD_BIGTREE_SKR_V1_3 has been renamed BOARD_BTT_SKR_V1_3. Please update your configuration."

ellensp, 2020-02-29:
That is after the fact, you have already hit compile and wasted your time. 
How many 1000's of people has this irritated when there was zero need for it? 
It was slipped in as it was added in a completely unrelated PR without any documentation or even hits that it was there.

ellensp, 2020-02-29:
There is also no reason the code had to be so "binary". It could easily have been coded to accept both board names without any difficulty.

thisismike97, 2020-02-29:
> There is also no reason the code had to be so "binary". It could easily have been coded to accept both board names without any difficulty.

Supporting both board names would create confusion, many would ask what the difference is.

> That is after the fact, you have already hit compile and wasted your time.

Compiling only takes a few minutes. There is so much else you could do while waiting for it, in my opinion it is by no means wasted time.

> It was slipped in as it was added in a completely unrelated PR without any documentation or even hints that it was there.

My mistake, you are correct on this. It should have had it's own commit.

sbaeder, 2020-02-29:
I agree that there is a sanity check *IF* you can compile it.  If you used the AutoBuild system, it just doesn't know how to build it!  Yes, tere are alternatives, and I'm glad to see that you agree it should have been in the release notes, separate commit, etc....
My intent wasn't to point fingers, but make comments to improve things in the future, and to me, that was accomplished.  I think that the AutoBuiild systems is a great *work-in-progress" that will eventually benefit a lot of folks!
ThanksScott

sbaeder@aol.com
508-331-1530 (cell)

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle