MarlinFirmware/Marlin #17024: Neopixel PRINTER_EVENT_LEDS not working Marlin 2.0.4.4

Testmangh, 2020-02-29:
### Bug Description

When the printer starts the neopixel LED strip (WS2812B) cycles trough RED, GREEN and BLUE but then it turns off even with PRINTER_EVENT_LEDS enabled. When setting the nozzle/bed temp to 80/40 the LEDs stay off.

### My Configurations

https://drive.google.com/drive/folders/1ynJWbr6kv8dwqPYqKGsg2E7YwMGrdWOi?usp=sharing

### Steps to Reproduce

With the configuration above turn on the printer, then set the nozzle/bed temperature to a value higher then the ambiente temperature.
https://www.youtube.com/watch?v=2UiLjbQ4c4g

#### Additional Information

SKR 1.3 LED Strip connected to the ZMAX Signal pin... I have unsoldered the C10 and R13 and shorted R14.
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-SCH.pdf

ellensp, 2020-03-01:
It does seem to be broken...    NEOPIXEL_STARTUP_TEST does the cycle of colours on startup.  M150 still works also...  Its just the PRINTER_EVENT_LEDS seems to do nothing.
I'm also on a BTT skr 1.3  but on a different port.
Using current todays bugfix 2.0.x 
```
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_RGB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN       EXPA1_05_PIN     // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 3       // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif
```

Moyster, 2020-03-01:
> PRINTER_EVENT_LEDS seems to do nothing.

what do you mean by that ? Led(s) can be turned on/off at printers' boot, but P_E_LEDS does nothing when you pre-heat your bed/nozzle manually, it works as intended when starting a (full) print (at least on my RGB_LED setup).  

Or was it working before and now it only cycle colors on boot and never does/light anymore after ?

ellensp, 2020-03-01:
It used to do this, as it is commented in Configuration.h 

/**
 * Printer Event LEDs
 *
 * During printing, the LEDs will reflect the printer status:
 *
 *  - Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface
 *  - Change to green once print has finished
 *  - Turn off after the print has finished and the user has pushed a button
 */

Moyster, 2020-03-01:
The feature still works here, this might be specific to neopixels' (code) ?

ellensp, 2020-03-01:
I don't believe so, If that was the case the colour test on power on and M150 also wouldn't work.

ellensp, 2020-03-01:
I just tried 3 standard LEDS, and changed to #define RGB_LED from #define NEOPIXEL_LED and setup the 3 pins.  Same results. All are controllable with M150 but when you tell the board to heat up the hot end, it should turn on red and blue leds for violet. Slowly reducing blue as hot end comes to temperature. It doesn't turn on anything.

ellensp, 2020-03-01:
Found something.
The status LEDS only display temperature indication when you use M109 not when you use M104
Similarly for the bed  M140 does not work but M190 does.

Moyster, 2020-03-01:
Oh, that might explain yep, Cura and PID autotunes always use M109/190 iirc.  
printer_event_leds has a few limitation also, the "green color" status is dependant on sdPrint.complete to know when prints are over, so printing over usb doesn't turn the led green.

Testmangh, 2020-03-05:
Guys sorry for disappearing, I thought I would receive an Email when someone answered the topic. 

> Oh, that might explain yep, Cura and PID autotunes always use M109/190 iirc.

Yeah someone on Facebook told me that it would only work with M109 and M190, after the printer starts it cycles trough RGB then off... Everything works perfectly, I think it is better for the white to not stay on...

> printer_event_leds has a few limitation also, the "green color" status is dependant on sdPrint.complete to know when prints are over, so printing over usb doesn't turn the led green.

Yeah it is exactly what is happening, I just put a end G-CODE on cura to tackle that: M150 R0 U50 B0.

Thanks guys and SORRY.

closed 2020-03-05 by Testmangh