MarlinFirmware/Marlin #17028: [FR] Maintenance menu

Sjoerrdd, 2020-02-29:
### Description
I would like too see a "maintenance menu" in the Marlin Firmware, to make it easier to maintenance your printer.
Things I would like to see:
- cold pull
- z-axis alignment fix
- z-probe live calibration
- load filament ("press knob when filament comes out of the nozzle" on ultimaker)

etc. (all the features that Prusa and Ultimaker implemented on their printers)

I know that it's already possible by running gcode, but it would be much more user-friendly if it's build-in. 
### Feature Workflow

1. Press the knob
2. Scroll to 'Maintenance'
3. Choose an option from the menu

#### Additional Information
[image]

[image]

Anthrix, 2020-02-29:
Sounds like good idea.

tvixen, 2020-03-01:
It's not a issue, but an idea. Use pull request.

SFRSKZ, 2020-03-01:
well you can add custom menus. try this one. it s almost same same with prusa firmware. credits to chris warkocki

https://github.com/codiac2600/SKR-MK3s-V1.4-Beta/blob/master/SKR%201.4%202.0.4.2/Configuration_adv.h

Sjoerrdd, 2020-03-12:
Ultimaker's "let plate cool down before removing print" screen would also be a nice addon to Marlin.

Sjoerrdd, 2020-04-12:
@SFRSKZ are you be able to upload it to a gist, since the repo is deleted?

thisiskeithb, 2020-04-12:
Here’s a link to the repo. Just drill down to find what you need: https://github.com/codiac2600/SKR-MK3s-V1.4-Beta

thinkyhead, 2020-04-12:
This feature is on the TODO list. Unfortunately most of our time is sunk into fielding issues and feature requests.