MarlinFirmware/Marlin #17030: Babystepping without effect

freaked1234, 2020-03-01:
After helping to resolve a bug with z probe temp. calibration yesterday maybe we manage to resolve another issue I just encountered....

### Bug Description

Babystepping doesnt work (anymore) - LCD menu shows up, but changing values doesnt do anything in either direction. (used as intended during first layer without pause)
It worked a few days ago though, before I finished setting up my probe and set up probe temp. calibration.

### My Configurations

Config files attached

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4271265/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4271266/Configuration_adv.txt)

### Steps to Reproduce

Use z-Babystepping as intended during first layer via LCD menu with my config.

HW: SKR 1.4, TMC2130 SPI

**Expected behavior:**

Movement of z-Axis on the fly

**Actual behavior:**

Digits in Display change, no movement of Z at all

#### Additional Information

Worked a few days ago, didnt change anything regarding the steppers. Only changes (I remember):
Finish setting up inductive Z Probe
Set up z safe homing
Calibrate z Probe temperature compensation

Tryed adjusting stepper current and disabled/enabled stealth chop.

DID NOT change min pulse setting yet since I am 100% positive it worked before, I didnt change it and I dont know exactly what it does.

bill-orange, 2020-03-01:
See issue 16828 go all the way to the end.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle