MarlinFirmware/Marlin #17031: [BUG] Sensorless homing on a COREXY with SKRPro 1.1/TMC2209 failed

Lino77, 2020-03-01:
#16420
Have exactly the same problem with SkrPro and tmc2209. G28 X works and at Y it only moves a few mm and stops at the wrong position G28 X G28 Y G28 Z works perfectly.
Marlin bugfix-2.0.x was 1.3.2020
#define HOME_Y_BEFORE_X
turn the prosess over, y will work, but x not
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4271551/Marlin.zip)

SFRSKZ, 2020-03-01:
+1 with skr v1.3 and 2130s. marlin 2.0.3 has this problem too. not always happening yet its there.

thisismike97, 2020-03-01:
Lower your "#define HOMING_FEEDRATE_XY" to (25*60) it increases the reliability and crashes less hard if it still goes wrong, however homing will go slower.

Tweak the feedrate in combination with your stallguard sensitivity to find the sweetspot, it works 99% of the time for me with 25*60 feedrate and a sensitivity of 70.

Lino77, 2020-03-01:
I have now tested several settings, from 22 * 60 - 50 * 60 feed rate and the error is still there, manual homing but G28 does not work

SFRSKZ, 2020-03-02:
normal currents and homing currents are same or different?  maybe homing current is causing to  trigger earlier(if only current is too low for your stepper motor)

Lino77, 2020-03-02:
I've already tried with current up to 1000 mAh no change

hliebscher, 2020-03-04:
for me with 650 mA works fine

Lino77, 2020-03-07:
okay found a fault, it was the stepper motor, exchanged and it worked

closed 2020-03-07 by Lino77