MarlinFirmware/Marlin #17039: [BUG] System Fan icon not animated

rflulling, 2020-03-01:
### Bug Description

System Fan Icon doesn't do anything. I rechecked the configuration and configuration_adv, as well as the flipped code in the SKR PRO pins file that lets my fan work. I have it set if any driver is used the system fan will come on and it does. But there is no representation. It's trivial I know. The other fans are all driven by the power supply and have no automation, they are always on. 

Compiles on Visual Studio and Platform IO with full updates. Marlin 2.0.3
with Intel Gen9-i5 & 16GB of RAM, 1TB free over SATA3. Running Windows 7 Pro. 
Compiled for BTT SKR PRO 1.1

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4272560/Configuration.zip)

### Steps to Reproduce

Compile Firmware with fan set to be animated. 
Install Firmware
Start machine
Move axis

**Expected behavior:** [What you expect to happen]
Icon should do, something, blink, rotate, something. 

**Actual behavior:** [What actually happens]
Icon is frozen. Never moves. Never blinks. 

#### Additional Information

Sure I could add a picture. But really? I think this one is pretty strait forward.

thinkyhead, 2020-03-02:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2020-03-02:
Perhaps you modified your pins file. When building with `bugfix-2.0.x` you may see this:
```
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1457:4: error: #error "You cannot set CONTROLLER_FAN_PIN equal to FAN_PIN."
   #error "You cannot set CONTROLLER_FAN_PIN equal to FAN_PIN."
```

rflulling, 2020-03-02:
Ok running a merge on the configuration now. Will reply again with an
update.

rflulling, 2020-03-02:
> `Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1457:4: error: #error "You cannot set CONTROLLER_FAN_PIN equal to FAN_PIN."`
>`    #error "You cannot set CONTROLLER_FAN_PIN equal to FAN_PIN."`

No, for sure not. This one gave me issues when I first started learning bugfix 2.x.x and VS. 
But the error was specific and it's stopped the process. 

The pin flip switches a port as a work around to this error. 

System Fan is physically installed on FAN0. 
However I was unable to get FAN0 to turn on as system fan. 

MyFix:
```
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN **PC8**           // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 60             // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255           // 255 == full speed
  //#define CONTROLLERFAN_SPEED_Z_ONLY 127  // Reduce noise on machines that keep Z enabled
#endif
```

also I switch PC8 and PC6 in the pins file. So the software thinks its activating Fan2 but is really activating Fan0. This is a work that worked by trial and error, for lack of better advice on the setup. 

-----

Trying to compile bugfix as of 03/01/2020, and I am getting new never before seen errors in areas where there have been changes in the configuration. So I will have to troubleshoot that before I can get back to the fan issue.

rflulling, 2020-03-02:
Ok, had to enable all of the Drivers in order for Dual Y to work. 

I have never needed to define more than Y2 as a driver. I guess I now also need E2 defined of Sanity check throws a major fit about Y2 not being defined and no extra slots being available. 
I also got an error about undefined heater pins but that went away. 

---

The fan ICON. Still stuck. 

With the number of changes happening. I guess its bound to break my own configuration. But the question. If it my configuration or something deeper? I get no errors regarding the fans or the icons at compile.

ellensp, 2020-03-02:
I tried your configs (as much as possable) but on my lpc1768 (disabled 2nd extruder and dual Y)
m106 S255 turns on fan and starts animation.
m107 stops fan and stops animation.

martend, 2020-03-02:
I tested it on a SKR 1.4 with TMC2208 and the dual TFT35 V3.0 also from BTT.
Marlin firmware 2.0.4.4 was compiled on a Linux box, I7 16GB memory, using PlatformIO for VS.
I bumped the number of bitmaps in Configurations_adv.h to 4 for the fan, just to make sure It would compile different then the version already running on the board.

In the LCD modus I see the fan spinning animation, in the touchscreen mode, using the material theme, the fan does not spin.

rflulling, 2020-03-02:
> 
> 
> I tried your configs (as much as possable) but on my lpc1768 (disabled 2nd extruder and dual Y)
> m106 S255 turns on fan and starts animation.
> m107 stops fan and stops animation.
> Added E1 back in, fan still animates...

Thank you for testing the configuration.

rflulling, 2020-03-02:
@martend 
> In the LCD modus I see the fan spinning animation, in the touchscreen mode, using the material theme, the fan does not spin.

Your observation is interesting. 
- I do not use a touch screen and have not changed that configuration. Unless something was updated in the background I don't know about that would create this effect?

rflulling, 2020-03-02:
**Summary for any one who doesn't like to read everything.** 

@ellensp you have hit on something interesting.
-System fan instruct the controller to turn on the fan when any Driver is used, and it does, and it switches off without issue on it's own. But the ICON does not animate. 
-You have found that using the "M106 S255" command does trigger the Fan and the ICON does spin. So this is useful in troubleshooting the code I think. 

@martend what you found was also useful and interesting. You noticed that for your set up: 
-the LCD showed fine.
-but the touch screen did not function. 

@thinkyhead Does the above info help?

ellensp, 2020-03-02:
fairly sure the fan icon is only for the extruder part cooling fan. Autofans for extruder cooling and System cooling fans are not showing in this icon.

martend, 2020-03-02:
@rflulling I think the fan is not animated in the touchscreen / material layout. I only saw one .bmp in the folder for it.

rflulling, 2020-03-02:
> fairly sure the fan icon is only for the extruder part cooling fan. Autofans for extruder cooling and System cooling fans are not showing in this icon.

If True. Then something has changed. The only reason for generating this ticket is because from 1.9 to 2.0.0 and all the bug fixes between. I have over 14 iterations. This worked.

rflulling, 2020-03-02:
@ellensp 
There may well be some truth to what you say. Though I will swear this change just happened. 

I am noticing that although I thought the system fan cam on earlier, it did not. The fan ICON has been on all this time. Nothing at all is assigned to the ports that are active. The system fan is not active. 

So the change that has happened was to make it so that ONLY the fans on the print heads will trigger the animation of the fan on the LCD. 

However since my fans are all tied direct this offers me no benefit. 

The question for Thinkyhead, is this change permanent?
I will not ask that the code be changed for just myself alone, especially if what I have grown used to was in itself a bug that has been fixed. Perhaps I might place a feature request where a user has the ability to determine which fan is tied to the animation?

martend, 2020-03-04:
@rflulling @ellensp @rflulling 
I did some further testing with printing from an SD Card and via Octoprint, in both cases the LCD emulation shows what it should, animation of the fan and the percentage it's blowing at. In touchscreen mode the fan is blowing, but the percentage stays at 0. 
How can I check if that issue is inside the firmware of the touchscreen or in Marlin?

thinkyhead, 2020-03-07:
> the change that has happened was to make it so that ONLY the fans on the print heads will trigger the animation of the fan on the LCD

The icon on the status screen has always represented the layer cooling fan, and no other fans have ever been displayed on the Marlin status screen (with icons). Some forks of Marlin do their own thing.

> How can I check if that issue is inside the firmware of the touchscreen or in Marlin?

We'd have to determine what method the screen is using to ask for the current fan speed. Marlin draws the fan speed itself on regular LCDs but in touch mode the LCD would need to send G-code to Marlin to ask for the fan speed, or if it's implemented under ExtUI it has to implement that part of the API.

---

I've migrated your configurations for the current Marlin `bugfix-2.0.x` branch if you feel like moving up to the most recent codebase:  [config-204.zip](https://github.com/MarlinFirmware/Marlin/files/4301316/config-204.zip)

rflulling, 2020-05-04:
I am closing this issue as its clear that marlin, on this topic is functioning as intended. While I swear it's operated different. I wont question it further. I see the intent established by Marlin and have no further reason to question the operation at this time.

closed 2020-05-04 by rflulling