MarlinFirmware/Marlin #17050: [BUG] UBL grid points visible in prints caused by LA pressure release

rado79, 2020-03-02:
### Bug Description

Printing with UBL and LA enabled shows visible artifacts on meshpoints. If nozzle travels over a meshpoint while priming, LA does a pressure-release-move on extruder. I noticed it a few weeks ago while doing some bed calibration prints with one layer. Yesterday i started to print some kind of curved lithophanes with translucent filament. They cover two or more of the meshpoints. I can see where the meshpoints are through the whole hight of the print. 
If you put any object placed on a meshpoint with outer walls , you can see vertical artifact on skin.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4275475/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4275476/Configuration_adv.zip)

### Steps to Reproduce

1. activate UBL and LA
2. place object to cover meshpoints
3. start print
4. watch extrudermoves while crossing meshpoints

**Expected behavior:** print as expected

**Actual behavior:** artifacts on meshpoints

#### Additional Information

9x9 grid, meshinset 30.
I tried various slicers and changing the position of the object in x and y direction. Artifacts are always on same coordinates.
First time i noticed is about 5 weeks ago, i download new bugfix once a week.
https://github.com/MarlinFirmware/Marlin/issues/10171#issue-307002075
https://github.com/MarlinFirmware/Marlin/issues/16910#issue-568377262

AnHardt, 2020-03-02:
Most likely you are not seeing the Artefakts at the mesh-points, but at the mesh-borders - the lines connecting the mesh-points.
And you don't see them at all of them, but only on those where z-speed crosses zero - z-direction changes.
Because all axes are synchronized, all axes including E  have to be slow there. That needs speeding down to the border and speeding up thereafter.
LAs job is to react on speed changes, dos it and lowers/ups the pressure accordingly.
These artifacts a unavoidable when all systems are doing their job. 
But with tuning the values for JD, E-jerk and LA they can be more or less visible.

Where lithophanes could be the use-case the problem is best visible.

Since lithophanes have a large contact area to the bed and therefor should stick well, i'd try to print them without or with linear BL. If that does not work - try a less bad bed.

rado79, 2020-03-02:
@AnHardt  First thank you for the good explanation. 
Yes you are right - seems to be mesh-borders not points. One thing i do not understand. I thought after passing fade height  G29 F 10.0 there is no z-move (layer change excluded ). I do my lithophanes vertical, and the artifacts are still visible at 120mm hight.  Shouldn´t the artifacts disappear over fade hight?

I work with bed leveling because of observing babystep issues https://github.com/MarlinFirmware/Marlin/issues/16617#, otherwise my bed is good enough to work without UBL.

Roxy-3D, 2020-03-03:
I can add some light to this discussion...   

When UBL was written, doing mesh based leveling added a lot of floating point math to every line segment printed.   And the line segments needed to be cut off at each mesh cell boundary.   UBL's Z-Height corrections fall on mesh cell boundaries.   All of the different cases for crossing those boundaries are broken out in UBL's motion generation code and highly optimized to eleminate unneeded floating point operations.

It may be time to jettison that code and just use the normal Planner:: code.    I think what happened is the Linear Advance people hooked in to the Planner:: and Stepper:: code and never bothered to update the UBL code to be compatible with Linear Advance.

rado79, 2020-03-04:
Any idea why artifacts are still visible above fade hight? There is no Z move or direction change except layer change, or am i wrong?

Roxy-3D, 2020-03-05:
I don't think what you are seeing is a 'bed leveling' problem per se.    The problem is all line segments are broken into more line segments at mesh cell boundaries.   And if the nozzle is below the fade height, some amount of Z correction is added (or subtracted) at that point.

The problem is Linear Advance is doing some thing extra at the mesh cell boundaries.     It may think a sharp turn is being taken (when it is not) or something else.   I don't know.

But we have two options...  We can hook UBL into the Stepper:: and Planner:: code and remove the optimized motion algorithms.   Or we can figure out what the incompatibility is and add what is needed to the UBL motion code.

rado79, 2020-03-06:
> Most likely you are not seeing the Artefakts at the mesh-points, but at the mesh-borders - the lines connecting the mesh-points.

In my case definitely emerging at the mesh vertices 
[image]
thank you @halninekilo for sharing picture

not like in this video: https://github.com/MarlinFirmware/Marlin/issues/10171#issuecomment-374865261

rado79, 2020-03-28:
Today i noticed a strange behavior. Because of bigger prints (about 320x320mm) i activated UBL. 
If printhead crosses the meshpoints, Z axis jumps up an down fast (a few 1/10mm, or in other words z-axis rotation about 10 degrees on TR8x8) building these artifacts. At the same time LA does pressure release (extruder back and forth). 

Increasing line width (0.48mm) and height (0.3mm) makes this behavior more visible. Normally i print with 0.15-0.20mm height. Latest bugfix installed once a week. 
[image]
[image]

jbaribeault, 2020-06-08:
I'm definitely seeing it too; first time I've noticed but I've only recently turned on LA and was doing a larger functional print at .28 layer height with .48 line width. Very visible in this particular filament.

[image]

boelle, 2020-06-21:
@rado79 still an issue?

rado79, 2020-06-21:
@boelle nice that you're back!

Today i tested again with latest bugfix, UBL 10x10 and LA enabled. This issue seems to be gone.
[image]

closed 2020-06-21 by rado79