MarlinFirmware/Marlin #17053: [BUG] (Marlin 2.0 BugFix Compilation)

RawdeHeho, 2020-03-02:
Hello, i new to github but use 3d printing and marlin for a few times now

I have some trouble to made compilation of marlin 2.0 bugfix (Marlin 1.1.x with bugfix configuration files) 
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4276688/Configuration_adv.zip)

with Arduino 1.8.10 and 1.8.12.

I need to precise that it work correctly with Marlin 1.1.9.

I have modify my 3D printer with MKS GEN L V2 board with adding TMC2209, but compilation doesn't work with marlin 2.0 and i don't know why.

Maybe someone can help me, i have try during hours to find on web what is wrong without success

Thanks you in advance

Here is my configuration files (in joigned file) and error message i get during compilation

I have try to made the change/modification asked by error message but it doesn't work and it only give me other errors message whatever i done and correct it

> 

Arduino : 1.8.10 (Windows 10), Carte : "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Erreur durant le téléchargement de https://downloads.arduino.cc/packages/package_index.json
In file included from sketch\src\inc/MarlinConfig.h:39:0,

Plusieurs bibliothèque trouvées pour "LiquidCrystal.h"
                 from sketch\src\MarlinCore.h:24,

Utilisé : F:\Programme
                 from sketch\src\MarlinCore.cpp:31:

c:\users\a001629\appdata\local\temp\arduino_build_500621\sketch\src\hal\hal_avr\inc\sanitycheck.h:64:4: error: #error "MONITOR_DRIVER_STATUS causes performance issues when used with SoftwareSerial-connected drivers. Disable MONITOR_DRIVER_STATUS or use hardware serial to continue."

   #error "MONITOR_DRIVER_STATUS causes performance issues when used with SoftwareSerial-connected drivers. Disable MONITOR_DRIVER_STATUS or use hardware serial to continue."

    ^~~~~

In file included from sketch\src\MarlinCore.cpp:37:0:

stepper.h:47:10: error: stepper/indirection.h: No such file or directory

 #include "stepper/indirection.h"

          ^~~~~~~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1
stepper/indirection.h: No such file or directory

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

thisiskeithb, 2020-03-02:
> #error "MONITOR_DRIVER_STATUS causes performance issues when used with SoftwareSerial-connected drivers. Disable MONITOR_DRIVER_STATUS or use hardware serial to continue."

Do as the error message says and disable `MONITOR_DRIVER_STATUS`.

This is not a bug.

sjasonsmith, 2020-03-02:
Older versions of TMCStepper simply ignored MONITOR_DRIVER_STATUS if you used software serial. It is now possible, but caused enough performance issues on AVR that the sanity check was added to educate people to avoid it.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle