MarlinFirmware/Marlin #17057: [BUG] Sensorless homing with SKR 1.3/TMC2209 failed

gsxr2001, 2020-03-03:
### Bug Description

Marlin 2.0.4.4 on SKR 1.3 with TMC2209 UART mode, sensorless home for X and Y, 3DTouch for Z. QUICK_HOME enabled

If I home axes individually, all work perfectly. If I home all (just G28) it goes just for 2 or 3mm. Then goes to opposide side (see video)

Got back to 2.0.3 all work again

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4280368/Configuration.zip)

### Steps to Reproduce

1. Go home with munù or G28 command

**Expected behavior:** [What you expect to happen]

It goes home

**Actual behavior:** [What actually happens]

It goes to opposide side

#### Additional Information

https://youtu.be/rguBzUyfv5Y

gsxr2001, 2020-03-03:
Disabling QUICK_HOME it works fine

drewzh, 2020-03-03:
Same experience here. Glad I'm not the only one - I've been diffing files trying to some configuration schema change but came up with nothing. Disabling QUICK_HOME in the meantime. Thanks for the heads up @gsxr2001

nklrk, 2020-03-04:
Also have the exactly same issue. I will try to figure out what the problem is. I have an backup of the firmware (~2 weeks) which doesn't cause this problem, unfortunately only the compiled bin file. But I guess I will find the revision by file-date.

cropduster, 2020-03-07:
I also have issues with sensorless homing on my SKR v1.2. It sometimes does not jome correctly after a ptint has finished andd another print is about to start. This seems to be random. In one case it didn't move any axis and hence 0 for x was where the extruder just happened to be and the print started in the right corner of the bed.

onek24, 2020-03-07:
Same issue with SKR mini e3. With QUICK_HOME enabled, he does auto homing in the opposite direction.

brettvitaz, 2020-03-07:
wow...

thinkyhead, 2020-03-08:
> Good thing I keep the working copy around...

Indeed. Would you send it over for comparison?

thinkyhead, 2020-03-08:
> Occurs with Marlin 2.0.4.4
> Last working version is Marlin 2.0.4.3

Ah, well we can just run a comparison on these then.

thinkyhead, 2020-03-08:
There are only three changes between 2.0.4.3 and 2.0.4.4 that seem likely to affect sensorless homing. (I have not looked at changes in the same time period in the `TMCStepper` library, but let's assume it's fine for now.)

1. These lines were added to `G28.cpp`. Try removing them:
    ```cpp
    do_blocking_move_to_xy(-0.5 * x_axis_home_dir, -0.5 * home_dir(Y_AXIS), fr_mm_s / 2);
    safe_delay(100);
    ```

2. A bug fix changed your `MINIMUM_STEPPER_PULSE` to 0 if you have Trinamics. The (assumed wrong) previous value was 2. Try setting `MINIMUM_STEPPER_PULSE` to 2 and see if it has any effect.

3. A bug fix changed your `MAXIMUM_STEPPER_RATE` to 5000000 if you have Trinamics. The (assumed wrong) previous value was 250000. Try setting `MAXIMUM_STEPPER_RATE` to some lower value like 2500000 … 1000000 … 500000 … or 250000 and see if this has any effect.

Thanks for helping to improve Marlin and to wrangle these interesting new hardware combinations!

onek24, 2020-03-08:
> 
> 
> There are only three changes between 2.0.4.3 and 2.0.4.4 that seem likely to affect sensorless homing. (I have not looked at changes in the same time period in the `TMCStepper` library, but let's assume it's fine for now.)
> 
>     1. These lines were added to `G28.cpp` (lines 109-110). Try removing them:
>        ```c++
>        do_blocking_move_to_xy(-0.5 * x_axis_home_dir, -0.5 * home_dir(Y_AXIS), fr_mm_s / 2);
>        safe_delay(100);
>        ```
> 
>     2. A bug fix changed your `MINIMUM_STEPPER_PULSE` to 0 if you have Trinamics. The (assumed wrong) previous value was 2. Try setting `MINIMUM_STEPPER_PULSE` to 2 or 1 and see if it has any effect.
> 
>     3. A bug fix changed your `MAXIMUM_STEPPER_RATE` to 5000000 if you have Trinamics. The (assumed wrong) previous value was 250000. Try setting `MAXIMUM_STEPPER_RATE` to some lower value like 2500000 … 1000000 … 500000 … or 250000 and see if this has any effect.
> 
> 
> Thanks for helping to improve Marlin and to wrangle these interesting new hardware combinations!

Uncommenting lines 109-110 in `G28.cpp ` definitely solves this issue for me. 
Therefore this merge https://github.com/MarlinFirmware/Marlin/pull/16872 broke it.

gsxr2001, 2020-03-08:
> > There are only three changes between 2.0.4.3 and 2.0.4.4 that seem likely to affect sensorless homing. (I have not looked at changes in the same time period in the `TMCStepper` library, but let's assume it's fine for now.)
> > ```
> > 1. These lines were added to `G28.cpp` (lines 109-110). Try removing them:
> >    ```c++
> >    do_blocking_move_to_xy(-0.5 * x_axis_home_dir, -0.5 * home_dir(Y_AXIS), fr_mm_s / 2);
> >    safe_delay(100);
> >    ```
> > 
> > 2. A bug fix changed your `MINIMUM_STEPPER_PULSE` to 0 if you have Trinamics. The (assumed wrong) previous value was 2. Try setting `MINIMUM_STEPPER_PULSE` to 2 or 1 and see if it has any effect.
> > 
> > 3. A bug fix changed your `MAXIMUM_STEPPER_RATE` to 5000000 if you have Trinamics. The (assumed wrong) previous value was 250000. Try setting `MAXIMUM_STEPPER_RATE` to some lower value like 2500000 … 1000000 … 500000 … or 250000 and see if this has any effect.
> > ```
> > 
> > 
> > Thanks for helping to improve Marlin and to wrangle these interesting new hardware combinations!
> 
> Uncommenting lines 109-110 in `G28.cpp` definitely solves this issue for me.
> Therefore it seems like this merge #16872 broke it.

What are 109-110 lines? Because in Marlin 2.0.4.4 at that lines I don't have uncommentable lines

onek24, 2020-03-08:
> 
> 
> > > There are only three changes between 2.0.4.3 and 2.0.4.4 that seem likely to affect sensorless homing. (I have not looked at changes in the same time period in the `TMCStepper` library, but let's assume it's fine for now.)
> > > ```
> > > 1. These lines were added to `G28.cpp` (lines 109-110). Try removing them:
> > >    ```c++
> > >    do_blocking_move_to_xy(-0.5 * x_axis_home_dir, -0.5 * home_dir(Y_AXIS), fr_mm_s / 2);
> > >    safe_delay(100);
> > > ```
> > > 
> > > 
> > > 
> > > 1. A bug fix changed your `MINIMUM_STEPPER_PULSE` to 0 if you have Trinamics. The (assumed wrong) previous value was 2. Try setting `MINIMUM_STEPPER_PULSE` to 2 or 1 and see if it has any effect.
> > > 2. A bug fix changed your `MAXIMUM_STEPPER_RATE` to 5000000 if you have Trinamics. The (assumed wrong) previous value was 250000. Try setting `MAXIMUM_STEPPER_RATE` to some lower value like 2500000 … 1000000 … 500000 … or 250000 and see if this has any effect.
> > > 
> > > ```
> > > 
> > > 
> > > Thanks for helping to improve Marlin and to wrangle these interesting new hardware combinations!
> > > ```
> > 
> > 
> > Uncommenting lines 109-110 in `G28.cpp` definitely solves this issue for me.
> > Therefore it seems like this merge #16872 broke it.
> 
> What are 109-110 lines? Because in Marlin 2.0.4.4 at that lines I don't have uncommentable lines

I'm sorry, i wrote that wrong. I mean "commenting out lines 109 - 110". 
The blocking move in line 109 moves the X and Y axis too far back.

adamrb, 2020-03-10:
Also seeing this on an SKR 1.4 with TMC 2209's.

closed 2020-03-10 by thinkyhead

gsxr2001, 2020-03-11:
The problem has not been resolved! Not completely. Commenting on the 2 lines, there is no longer the problem of moving in the opposite direction, but the quick home still does not work! At this point it is better to disable QUICK_HOME. So the problem is not solved.

onek24, 2020-03-11:
> 
> 
> The problem has not been resolved! Not completely. Commenting on the 2 lines, there is no longer the problem of moving in the opposite direction, but the quick home still does not work! At this point it is better to disable QUICK_HOME. So the problem is not solved.

How does the Quick Home not work?

gsxr2001, 2020-03-11:
The problem of opposite direction can be solved commenting (disabling) QUICK_HOME. The result is the same

onek24, 2020-03-11:
> 
> 
> The problem of opposite direction can be solved commenting (disabling) QUICK_HOME. The result is the same

Yes because it disables this 2 Lines. Uninstalling Marlin will solve this issue too... duh

thinkyhead, 2020-03-11:
If `QUICK_HOME` doesn't work for your particular tuning of TMC drivers, it is recommended to disable it. For many setups it does work at this point.