MarlinFirmware/Marlin #17059: endstop works in software but not hardware

tbojim1975, 2020-03-03:
This may just be a simple fix. I have the Bigtreetech GTR v1.0 board and the standard micro stops that came with the creality ender 3. All seems to be triggering in prontierface software from open to trigger. But Im not sure what might be causing these endstops to not stop my zmin, xmin, ymin axes when triggered in pronterface? It shows its triggered but the axes never stop? Any ideas?
This is the last of my setup before i do my first Print.

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use one of the support links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

Before filing an issue be sure to test the "bugfix" branches to see whether the issue has been resolved.

-->

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-03-03:
If M119 correctly shows the endstops as Triggered when they are triggered and OPEN end they are not triggered. There are two main posabilites for the axis not stoppping. 
1) you have the wrong endstops plugged into the axis, so for eg you home X and the Y endstop gets triggered. It will ignore it.  
2) you have your axis all about face.  Eg if you X endstop is on the left, but you set home direction to MAX but reverse stepper direction to force it to move left, it will ignore it as it is looking a X-max end stop not X-min. 

None of these are Bugs in marlin, but simple configuration issues.

Did you not see the line 
> Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use one of the support links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

thisiskeithb, 2020-03-03:
If you have stallguard-capable drivers installed (TMC2130/2209/5160, etc.), you have to remove the diag pin or they’ll interfere with endstops/probes.

Take a look at page 15 in the [GTR user manual](https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/blob/master/BTT%20GTR%20V1.0/manual/GTR%20V1.0%20Operating%20Instruction.pdf).

tbojim1975, 2020-03-04:
I use 6 tmc2208 drivers in uart mode on all 6 steppers. In pronterface the
x min, y min, z min are showing the correct open and triggered when
pressed. I saw something "maybe" with the uart jumpers in the manual you
sent, of which thank you for that.

tbojim1975, 2020-03-05:
Thank you guys. After comple I never did a simple restart lol I have a few projects going at the same time is my excuse :-)

closed 2020-03-06 by thinkyhead