MarlinFirmware/Marlin #17060: [BUG] (can't use another temp sensor pin on MKS GEN L)

TheMath38, 2020-03-03:
Hi ! I just got my MKS Gen L board and i just figured out that my temp pin sensor (number 1) died. So I got an idea : use the other temp connector on my printer in order to not buy a new motherboard.

So I searched on the pin_RAMPS.h file and switch the pin of the sensor 1 to the sensor 2, just like this : 

//
// Temperature Sensors
//
#define TEMP_0_PIN         15   // Analog Input
#define TEMP_1_PIN         13   // Analog Input
#define TEMP_BED_PIN       14   // Analog Input

But... it definitely don't work, I still have "0" on my screen when I'm looking of the Extruder's temperature.
I searched a lot on forums, even here, without founding any solution, so now i'm opening an issue, believing that you could found a solution for me :3

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4281956/Marlin.zip)
I'm sending to you my configuration.h/pins.h/pins_RAMPS.h and configuration_adv.h file to investigate the "issue". Ho, and I forgotten to mention that I'm still on Marlin 1.1.9 (i don't really have to time to upgrade my firmware to 2.0 huuu shame on me)
Thanks in advance ! :)

borland1, 2020-03-03:
When I disconnect the thermistor wiring plug at the controller board of my MKS_GEN_L, the display says -25, which is in degrees C.   I tried looking at your uploaded zip file, but the zip folder is empty.  There are configuration.h settings for the thermistor type.  Did you check that also?

The MKS_GEN_L boards are now available in either V1.0 and V2.0.  Cost is basically the same and only about $16USD delivered to USA from China.  V2.0 supports serial TMC drivers without wiring mess.

ellensp, 2020-03-04:
Often when the a analog pin dies what really dies is the analog to digital converter. Ie all analog ports are dead.

TheMath38, 2020-03-04:
> When I disconnect the thermistor wiring plug at the controller board of my MKS_GEN_L, the display says -25, which is in degrees C. I tried looking at your uploaded zip file, but the zip folder is [...]

Really ? I rezip the file so hope this time it will work :3 
Yup, I checked the thermistor type (wich is 5 in my case) but I don't think that the problem is here, because even when I tried to change it (switch it from 5 to 60), it didn't change anything...
I hope that both connector pin are not died (that would be an huge shame!) 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4286371/Marlin.zip)
 Ho, and thanks for the reply :)

borland1, 2020-03-05:
Type 11 thermistor is what I have configured on my configuration.h file.  That gives a -25C display reading on display when thermistor sensor is disconnected at the controller board.

These processors are pretty robust, so I doubt that the ADC is defective on yours.  One could write a simple Arduino program to check what the ADC registers are measuring with fixed resistors across the board's terminals.

ellensp, 2020-03-06:
Or if they cant do that they can just use my tester https://github.com/ellensp/RAMPS_T0-T2_tester 
Uses same pins as ramps for thermistors.

TheMath38, 2020-03-06:
> Or if they cant do that they can just use my tester https://github.com/ellensp/RAMPS_T0-T2_tester
> Uses same pins as ramps for thermistors.

Hey ! :) So i tested your program on both connector (firstly on the T0 connector, and after on the T1 connector) ; Here is the results :
[image]
[image]

I think that both connector seems to be reading 0 (and weirdly 1023 on the other one?) even on the connector 1 or 2, but strangely the BED connector seems to be fine.... I think that I'm gonna buy a new motherboard because this problem is going to make me mad soon :333

ellensp, 2020-03-06:
To me this indicates there is no issue on the board, your thermistor is shorted thus you get zero on the analog pin.

thinkyhead, 2020-03-06:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-03-06 by thinkyhead