MarlinFirmware/Marlin #17062: [BUG] Anycubic Kossel Delta Plus, auto calibration fails

pablopeu, 2020-03-03:
Printer Anycubic Kossel Delta Plus, type 2 probe, marlin 2.044, modified provided example config.

Modified the bare minimum of the configuration example provided, when I start auto calibration it touches the table twice and fails. 
Using the paper method I entered the proper delta height, and then did the bed leveling with the probe (type 2) which finished successfully, now its printing fine.
So at least for me Marlin 2.044 works fine so far for everything but auto calibration.
Attached are the config files.
Thanks

sjasonsmith, 2020-03-04:
Try with the bugfix-2.0.x branch. Related issues have been fixed since 2.0.4

pablopeu, 2020-03-04:
Autocalibration worked fine with 2.04bugfix. Attached are the slighltly modified config files.

[Marlin 2.044bugfix 4-3-20.zip](https://github.com/MarlinFirmware/Marlin/files/4290087/Marlin.2.044bugfix.4-3-20.zip)

thinkyhead, 2020-03-06:
What a difference a half a mm makes!

pablopeu, 2020-03-06:
well, its not that, after resetting eeprom, doing a full g28>g29>m500 the bed is still not properly level, on the right is higher, on the left is lower, this is clearly seen on prints that cover almost all the delta bed. On the second layer, on the left side looks like a 1st layer and on the right side its as expected.

pablopeu, 2020-03-06:
[image]

closed 2020-03-14 by pablopeu