MarlinFirmware/Marlin #17065: [BUG] Marlin does not compile if I activate RepRapDiscount LCD

TabajaraLabs, 2020-03-04:
### Bug Description
(sorry, first bug report ever)
When I try to compile Marlin 2.0.x or bugfix 2.0.x version WITH RepRapDiscount graphic LCD option uncommented (activated) I get the following compilation errors: 
(The errors doesn't happen with the option commented (deactivated)

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp: In function 'int lcd_put_wchar_max(uint32_t, pixel_len_t)':

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:36:22: error: 'class U8GLIB_ST7920_128X64_RRD' has no member named 'getPrintCol'; did you mean 'setPrintPos'?

   u8g_uint_t x = u8g.getPrintCol(), y = u8g.getPrintRow(),

                      ^~~~~~~~~~~

                      setPrintPos

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:38:23: error: 'ret' was not declared in this scope

   u8g.setPrintPos(x + ret, y);

                       ^~~

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:38:23: note: suggested alternative: 'reti'

   u8g.setPrintPos(x + ret, y);

                       ^~~

                       reti

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:38:28: error: 'y' was not declared in this scope

   u8g.setPrintPos(x + ret, y);

                            ^

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp: In function 'int lcd_put_u8str_max(const char*, pixel_len_t)':

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:43:22: error: 'class U8GLIB_ST7920_128X64_RRD' has no member named 'getPrintCol'; did you mean 'setPrintPos'?

   u8g_uint_t x = u8g.getPrintCol(), y = u8g.getPrintRow(),

                      ^~~~~~~~~~~

                      setPrintPos

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:45:23: error: 'ret' was not declared in this scope

   u8g.setPrintPos(x + ret, y);

                       ^~~

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:45:23: note: suggested alternative: 'reti'

   u8g.setPrintPos(x + ret, y);

                       ^~~

                       reti

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:45:28: error: 'y' was not declared in this scope

   u8g.setPrintPos(x + ret, y);

                            ^

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp: In function 'int lcd_put_u8str_max_P(const char*, pixel_len_t)':

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:50:22: error: 'class U8GLIB_ST7920_128X64_RRD' has no member named 'getPrintCol'; did you mean 'setPrintPos'?

   u8g_uint_t x = u8g.getPrintCol(), y = u8g.getPrintRow(),

                      ^~~~~~~~~~~

                      setPrintPos

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:52:23: error: 'ret' was not declared in this scope

   u8g.setPrintPos(x + ret, y);

                       ^~~

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:52:23: note: suggested alternative: 'reti'

   u8g.setPrintPos(x + ret, y);

                       ^~~

                       reti

C:\Users\ALEXAN~1\AppData\Local\Temp\arduino_build_62591\sketch\src\lcd\dogm\lcdprint_u8g.cpp:52:28: error: 'y' was not declared in this scope

   u8g.setPrintPos(x + ret, y);

                            ^

"C:\\Users\\Alexandre\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Users\\Alexandre\\Documents\\Arduino\\libraries\\U8glib" "C:\\Users\\ALEXAN~1\\AppData\\Local\\Temp\\arduino_build_62591\\sketch\\src\\lcd\\dogm\\u8g_dev_st7920_128x64_HAL.cpp" -o "C:\\Users\\ALEXAN~1\\AppData\\Local\\Temp\\arduino_build_62591\\sketch\\src\\lcd\\dogm\\u8g_dev_st7920_128x64_HAL.cpp.o"
Using library U8glib in folder: C:\Users\Alexandre\Documents\Arduino\libraries\U8glib (legacy)
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

### My Configurations
[reprapdiscount error.zip](https://github.com/MarlinFirmware/Marlin/files/4287266/reprapdiscount.error.zip)

### Steps to Reproduce

1 - Alter the following line: 
// RepRapDiscount FULL GRAPHIC Smart Controller
// http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
//
//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

to

// RepRapDiscount FULL GRAPHIC Smart Controller
// http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
//
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

And in compilation I get these errors

**Expected behavior:** [What you expect to happen]

I expect to code to compile and upload to the arduino board

**Actual behavior:** [What actually happens]

it doesn't compile showing the above errors

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-03-04:
Current Bugfix and your config files build fine.
My guess is you using Arduino IDE and have old glib U8glib 
Please consider using vscode with platformio. It automatically gets any required libraries and produces smaller code. 
https://marlinfw.org/docs/basics/install_platformio_vscode.html

TabajaraLabs, 2020-03-04:
I got the most recent arduino compiler and the most recent U8glib...I don't believe I should install ANOTHER program to find the right lib. As an old timer, I find this arduino ecosystem very confuse. I gotta have the right compiler version, the right lib version...And now I gotta install another program to find the right libs to compile?! This is not the first time I have problems with this versioning problem...Anyways thanks for your advice, and here I go install another thing just to compile a code...oh my :(

ellensp, 2020-03-04:
The other option would be to delete the directory  "C:\Users\Alexandre\Documents\Arduino\libraries\U8glib (legacy)"
And try again...

ozzysv, 2020-03-04:
use vscode , when created arduino - no one could have thought that adults peoples  would create 3D printers based on it

TabajaraLabs, 2020-03-04:
Ellensp: Problem solved, thanks. 
It wasn't a "U8glib (legacy)" directory. It was another directory (sorry, didn't take the name, U8something) that was there. I BELIEVE that this other directory was created because I downloaded the U8glib and copied manually the library dir to arduino/libraries. 
Anyway, problem is solved and I thank you so much for the help!

closed 2020-03-04 by TabajaraLabs