MarlinFirmware/Marlin #17066: [BUG] Wrong temptable on configuration examples for Ender 3

VijitCoder, 2020-03-04:
I'll be short cause I tired fighting with printer after update to Marlin 2.x

\[Marlin-2.0.x/config/examples/Creality/Ender-3/Configuration.h]:

```
#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_BED 1
```

There should be "5" - ATC Semitec 104GT-2/104NT-4-R025H42G. 

I figured that out after several manual calibrations of the hotend thermistor. After all I built my own temptable and suddenly figured out that it matched with thermistor_5.h. 

So, I suggest to fix the example config for Ender 3.

thisiskeithb, 2020-03-04:
That should be 1. Why do you say they are 5?

VijitCoder, 2020-03-05:
Well, the point is that I didn't change thermistors. The printer assembled in Jan'2019.

If it does matter, I changed the motherboard to SKR. And uploaded Marlin 2.x there.

> They should be 1. Why do you say they are 5?

Because I built the temptable, as I said before. I used a thermometer and made 5 calibrations and a lot of concomitant checks.. After all I compared my results with the temptables in the Marlin sources. And it matched with thermistor_5.h.

As a beta test, I assembled a test stand, where I coupled the stock motherboard and the new one. In short: SKR board heats the hotend and uses its thermistor. From the stock board another thermistor attached to the same hotend. Repetier reads the stock board, also on the printer's LCD I see the temperature. And the thermometer is there, too, for monitoring.The result matched again.

Creality doesn't provide their sources, so I don't know what temptable they used. But the facts are as I have described. Maybe it's only with my printer, I admit it. Then maybe you can make your own tests? Can you?

> Could you submit a PR?

No, I can't cause I didn't get what happened with configurations examples. In the corresponding path I found only a link to archive. Download the _sources_ archive from the repo.. hmm, it is weird.

VijitCoder, 2020-03-05:
BTW: another thermistor that I used in the tests, was bought in Lerdge shop [here](https://www.aliexpress.com/item/32890886143.html). And it also matched with thermistor_5.h temptable. I can't say that it is exactly the same as that of Creality. I only say that the temperature measurements were the same for him.

VijitCoder, 2020-03-05:
@thisiskeithb In your screenshot said about CR-10 and CR-10S. **Ender 3** is a bit different printer. Cound you ask them about _that_ printer?

And as mentioned in the issue [#15039](https://github.com/MarlinFirmware/Marlin/issues/15039), the problem was also with CR-10, not Ender. And there before fixing was `#define TEMP_SENSOR_BED 5`. I can assume that Creality are mixing their printers configurations.

And I forget to say it before: I also checked the bed with thermistor_5.h using pyrometer. The temperature become much closer to the real measurements. Of course I can't heat the bed up to 250C, but even on 100C I saw, how it become better.

thisiskeithb, 2020-03-05:
Check the link. Creality sells the same NTC 100K/type 1 thermistor for both the CR & Ender series. This is not a bug.

VijitCoder, 2020-03-05:
Yeap, I see.

As you wish, you can close this issue without changes. I have nothing to add as a proof if all what I said above wasn't enough.

Good luck.

thinkyhead, 2020-03-06:
@VijitCoder — I'll make the patch for you. Do you suppose other Creality3D machines also use thermistor 5?

thinkyhead, 2020-03-06:
@thisiskeithb — Well, that is another point. The most common thermistors seem to be 1 and 5, and these show up on lots of machines, and it's not easy to tell which is which.

thisiskeithb, 2020-03-06:
> @thisiskeithb — Well, that is another point. The most common thermistors seem to be 1 and 5, and these show up on lots of machines, and it's not easy to tell which is which.

Adding an error block to force users to pick one or the other would work, but I still think 1 is correct for Creality machines since many share the exact same hardware.

VijitCoder, 2020-03-06:
> Do you suppose other Creality3D machines also use thermistor 5?

I donno :) I'm only a user with _one 3D printer_ and the temperature measurement tools.

I don't need the patch. I already found the right temptable and switched to it via configuration.

> Adding an error block to force users to pick one or the other would work

I'm not sure, what exactly you mean. Anyway, how the user can figure out what to choose without the temperature measurement? E.g. it was not clear for me, when I used the 1st table and it took ~4 days for digging around.

VijitCoder, 2020-03-06:
> I still think 1 is correct for Creality machines since many share the exact same hardware.

Can someone check this with another Ender? As an independent test.

I know that Creality makes changes to the model range and the same printers assembled in different years may differ. Maybe that's the root of the problem - they changed something?

swilkens, 2020-03-06:
Creality does provide their sources:
https://creality.com/download/source-code_c0001

https://github.com/Creality3DPrinting/Ender-3/blob/master/Ender-3%20Firmware%20(Marlin)/Ender-3%20(includes%20power%20failure%20resume-%20English)/Marlin/Configuration.h#L286

[image]

VijitCoder, 2020-03-06:
> Creality does provide their sources

I didn't know that.

I checked and confirm that in v.1.1.6 uses thermistor 1. 

At this point I have no idea why I've got a different result in all my tests including cross-test with two motherboards.

VijitCoder, 2020-03-06:
If no one else has any ideas, I suggest closing the issue.

swilkens, 2020-03-06:
> At this point I have no idea why I've got a different result in all my tests including cross-test with two motherboards. And why, after all, my _manually built temptable_ matched to 5, which is also mentioned in relation to Creality printers.

Where is this "5" mentioned in relation to creality printers?

Do you perhaps have a saintsmart branded ender 3?

That said, it wouldn't be the first time that Creality makes changes to a product without a new revision name.

thinkyhead, 2020-03-06:
I'll add a section to the Troubleshooting page that sometimes thermistors are changed on printer models without any notice from the manufacturer. I'll add a paragraph recommending to re-check the thermistor to see whether 1 or 5 (or something else) gives better results. I'd add a chart on how to recognize a thermistor, but unfortunately they can't really be seen when they're installed.

closed 2020-03-06 by thinkyhead

VijitCoder, 2020-03-07:
> Where is this "5" mentioned in relation to creality printers?

In another issue mentioned here - #15039, there was changed 5 to 1.

> Do you perhaps have a saintsmart branded ender 3?

What is "saintsmart"? As for my printer, I bought him in the official Creality shop on AliExpress in January 2019.