MarlinFirmware/Marlin #17073: [BUG] (GTR v1.0 CR-10 display)

Ski3d, 2020-03-05:
Hello,
      Just wanted to ask if anyone had a problem with the GTR and using the stock ender 3 display. I have CR10 display selected in the configuration.h I also have a 10 pin going from aux3 on the display to exp1 on the board. But I only get a blue screen(it lights up). Since this board is very new, I was wondering if the pins file was actually correct? I looked at the BTT pins file from there github and looked as if they had more enabled in there pins file. 
When I get home i can upload my config.h and configv.h file. But I was just throwing it out there if anyone else was having the same problem.

Ski3d, 2020-03-06:
welp found out you have to use exp1 on the display and flip the connector and then use REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER with the ender 3 screen that has exp 1,2,3

closed 2020-03-06 by Ski3d

thinkyhead, 2020-03-06:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).