MarlinFirmware/Marlin #17074: [BUG] Skr pro 1.1 not working onboard sdcard

Keltere, 2020-03-05:
After a lot of testing and trying different branches, settings etc... i couldn't find a solution so i decided to open this issue.
The problem is simple the onboard sdcard isn't initialized and used by the motherboard no matter what configuration is used.
For my testing config i just enabled `SDSUPPORT` and `TMC_SPI_SW` i kept this two consistent and then tried numerus thing like avoiding to declare the lcd serial or even removing the TMC_SPI_SW but nothing worked out, the sdcard won't mount no matter what.
I did try to use just the usb and then just the lcd but in neither one of the two cases i was able to mount the sdcard.
I did tried 3 different cards of different sizes two by Sandisk, but that wasn't the problem
With the LinoBarreca/Marlin/tree/MultiSPI_original the sdcard will be recognized and mounted but if unmounted it will not be hosted to be accessible to the device attached trough usb.
Another "solution" proposed by MikeEitel is to delete the check of the lcd spi on the pin declaration file like written [here](https://github.com/MarlinFirmware/Marlin/issues/16982#issuecomment-593135791) but even here when unmounted the sdcard will not be hosted.

TL;DR
1)Currently not working onboard sdcard on stable and bugfix branch
2)On LinoBarreca/Marlin/tree/MultiSPI_Original branch and Marlin/bugfix2.0 branch (removing the lcd check on pin file) the sdcard won't be hosted when unmounted

thisiskeithb, 2020-03-05:
This is a known issue and is a work in progress: https://github.com/MarlinFirmware/Marlin/pull/16260

Keltere, 2020-03-05:
> This is a known issue and is a work in progress: #16260

I did create this issue mainly for people in search of a solution but yeah, i should have linked the pull request.
Thanks.

Keltere, 2020-03-05:
The LCD check has been fixed but it still won't host the sdcard externally when unmounted and won't work if the lcd in SPI mode (exp1 and exp2 slot).
https://github.com/MarlinFirmware/Marlin/commit/e3da86fb452c1f8239033224976033d43853f4e3

AnHardt, 2020-03-05:
> won't host the sdcard externally when unmounted and won't work if the lcd in SPI mode

How could it. For the used core not any example exists for a USB-SD-card device. I will last a bit longer until that will work. (Guess: 1-2 Years) 
Don't wait for it.

MikeEitel, 2020-03-06:
Just to clarify that i understand it correct:
The skr pro is not enable to write on the onboard sd via usb as the mcu related software does not support this funtionality. Right? 

P. S.  This would be a real pitty as this is a nice board and can support 3 extruders without "addons"

thinkyhead, 2020-03-06:
> Right?

Yes.

closed 2020-03-06 by thinkyhead