MarlinFirmware/Marlin #17075: Printer not Returning to Height after Pause

JubilantReaper, 2020-03-05:
Current Configuration
Printers
CR-10 – 1.1.5 Board, Marlin 2.0.4.4, BLTouch
Ender 5 – 1.1.5 Board, Marlin 2.0.4.4, BLTouch
Octoprint – Each Printer has a Pi4, same builds, same Add-ons
Cura 4.5 – Use same computer to Slice, Same Settings
Use Cura and add PauseAtHeight
End 5 no issues, completed 6 plus files changing filament and returns print without issue.  
CR-10 – When the nozzle returns it’s about 1mm over the build surface, never returns to the proper height. Failed 4 times
Why I’m considering a Marlin Bug, both printers are using the same boards, same everything as much as possible. The delta is the Firmware Config for the printers. 
Snap shot of the GCODE for the item I’m printing where the pause takes place. Not being a coder, nothing is jumping out at me. 

G1 F4800 X127.474 Y138.107 E2651.877
G0 F9000 X127.936 Y138.228
G1 F4800 X135.473 Y145.766 E2652.0459
G1 F2700 E2647.0459
;MESH:NONMESH
G0 F300 X135.473 Y145.766 Z5
G0 F9000 X135.023 Y146.688
G0 X135.381 Y147.711
G0 X134.191 Y155.925
G0 X138.52 Y160.244
G0 X147.337 Y195.441
G0 X147.662 Y195.74
G0 X148.815 Y197.168
G0 X149.704 Y198.762
G0 X150.098 Y200.065
G0 X151.007 Y210.088
G0 X133.966 Y227.145
G0 X161.307 Y251.202
G0 X160.622 Y252.041
G0 X166.17 Y262.916
;TIME_ELAPSED:88805.058659
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 48
M83 ; switch to relative E values for any needed retraction
G1 F300 Z6 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z15 ; too close to bed--move to at least 15mm
M104 S0 ; standby temperature
M0 ; Do the actual pause
M109 S200 ; resume temperature
G1 F300 Z6
G1 F9000 X160.476 Y257.989
G1 F300 Z5 ; move back down to resume height
G1 F2700 ; restore extrusion feedrate
M82 ; switch back to absolute E values
G92 E2647.0459
;LAYER:48
;TYPE:WALL-INNER
;MESH:PerkinsHarley.stl
G1 F2700 E2652.0459
G1 F2400 X160.475 Y257.989 E2652.16522
G1 X152.943 Y251.837 E2652.31932
G1 X149.255 Y249.004 E2652.393
G1 X144.906 Y245.758 E2652.47899

thinkyhead, 2020-03-06:
You might want to try lowering your Z acceleration as a test. The code seems fine and I see nothing in Marlin itself that would cause Z moves to behave unusually under any conditions, whether there's been an `M0` or not.

Anyway, the more appropriate command to use would be `M125` which will also do parking behavior when you have the `PARK_HEAD_ON_PAUSE` option enabled. (When pausing an SD print from the host you would use `M25`.)

JubilantReaper, 2020-03-06:
thinkyhead, 
Thanks for the response... 

Z Axis, it's on a leadscrew and everything appears to solid, I do not see any slop, play, moves easily and everything is tight. Backlash on Z seems unlikely, layers are consistent and given the delta at least 1mm at the time of change, it's my opinion there is a software issue somewhere.

The appropriate commends, these were generated by Cura, I have not modified the code from what was produced. I did not pause, but used the the plugin's from cura. 

If I can rule our mechanical issues with the printer, the only delta between the Ender-5 and the CR-10 is the configuration and configuration_adv files. 

I have performed similar tests at different layer levels and all produce the same results with the CR-10, Ender 5 has no issues and works perfectly. 

Is there some logs I can try to capture to identify any issues?

thinkyhead, 2020-03-07:
For some possibly useful info, enable `M114_DETAIL` and then put `M114 D` in between all the commands before and after the `M0` where Z gets lost. Also insert some Z moves that don't involve any `M0` for contrast.

JubilantReaper, 2020-03-14:
thinkyhead, sorry for the delay, I have been out of pocket for the last week. I will run commands and see what I get. Thanks...

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle