MarlinFirmware/Marlin #17076: [BUG] Anycubic Kossel Delta Plus Bed leveling not OK

pablopeu, 2020-03-06:
After installing 2.04bugfix solved my autocalibration problem, but after further testing I noticed that bed leveling is not working as expected. If I print on the center of the bed the problem is unnoticeable, but then I printed many pieces distributed over the bed and the problem was obvious, to the left the head presses over the bed, in the center is level and on the right is floating by almost 1mm.

I'm using the provided touch probe which is attached magnetically below the extruder end and both autocalibration and bed leveling end without errors and values are stored.

Here is an image of the probe

[image]

And the config files I'm using:

[Marlin 2.044bugfix 4-3-20.zip](https://github.com/MarlinFirmware/Marlin/files/4298317/Marlin.2.044bugfix.4-3-20.zip)

thinkyhead, 2020-03-06:
Have you run `G33` calibration yet?

pablopeu, 2020-03-06:
I'm redoing the whole setup as I write this after erasing eeprom. Will report later.

thinkyhead, 2020-03-06:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-03-06 by thinkyhead

freiform, 2020-03-08:
> I'm redoing the whole setup as I write this after erasing eeprom. Will report later.

After folliwng the documentation on levelling and calibration to the letter, I am having the exact same issue with bugfix-2.0.4. Any updates?

thinkyhead, 2020-03-08:
If you can find a version of Marlin where the kinematics work perfectly for you, please bring that information to us.

pablopeu, 2020-03-09:
Downloaded latest bugfix a couple of hours ago, config files are the same as uploaded earlier, did a full eeprom init, delta calibration (from the lcd menu) fails after some touches on the bed. Then I tried to adjust the printer height using the paper method. saved the values, and sent a print, it was tall, by about a mm, so then I tried auto calibration again (from the lcd menu) and it worked fine, weird. Did the paper calibration again just in case, slightly corrected the height, and stored settings. 
Now its printing a 15hs job, will check it tomorrow.