MarlinFirmware/Marlin #17077: [BUG] (short description) print '\nUnable to find destination disk (' + e + ')\n' \           'Please select it in platformio.ini using the upload_port keyword ' \           '(https://docs.platformio.org/en/latest/projectconf/section_env_upload.html)\n' \           'or copy the firmware (.pioenvs/' + env.get('PIOENV') + '/firmware.bin) manually to the appropriate disk\n'

vixa80, 2020-03-06:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

vixa80, 2020-03-06:
Good afternoon. I’m trying to install Marlin 2.0 Visual Studi Code. But I get this error. In programming, I don’t understand Piton, I don’t understand either. Explain to me what and where to do it please.

ellensp, 2020-03-06:
Can't follow instructions either...  

> Required: Please include a ZIP file containing your Configuration.h and Configuration_adv.h files.

The code compiled without issues. 

The error says it can't find the SD card mounted as a drive on your operating system to copy the new firmware.bin file to. 

Firstly check that when you plug your controller into you computer with the SD card in the controller your operating system sees SD card as a drive.   

To proceed from here we need to know what board you are building for, which would have been in the requested Configuration files.....

Some boards look for a file on the SD card FIRMWARE.CUR to identify where to upload the new firmware.bin  (file can contain anything, just the file name is important) 
Some look for a SD card volume name of "REARM" (lpc1768/9 based boards) 
Some boards are happy either either of these two methods.
Or it could be something else... Depends on what board your building for.

closed 2020-03-06 by thinkyhead

b7-code, 2020-03-07:
hello guys, I'm new GitHub and i would like to seek some help in my project. 
Im building a scara 3d printer using the MKS GEN L V2.0 board and marlin 2.0.1

the problems I'm facing is that after uploading the firmware the machine only homes the z axis, any other command it freezes the sends the error of printer killed. 
any help would be appreciated

thinkyhead, 2020-03-08:
@b7-code — This issue queue is a tool for developers to keep track of our bugs and we don't want to do support here. Please use one of the following options to get help from the community:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

b7-code, 2020-03-08:
@thinkyhead thank you for the information, I’ll check them out 👍