MarlinFirmware/Marlin #17078: Bed PID values read from EEPROM aren't used if P is equal to DUMMY_PID_VALUE

Alfrederson, 2020-03-06:
Took quite a while to find out why my friend's controller wouldn't use his PID values no matter what. Turns out he was using 3000 for his P, and configuration_store.cpp is written in a way that it will only update the thermal manager's values if P isn't the same value as DUMMY_PID_VALUE.

Check:

```
      //
      // Heated Bed PID
      //
      {
        PID_t pid;
        EEPROM_READ(pid);
        #if ENABLED(PIDTEMPBED)
          if (!validating && pid.Kp != DUMMY_PID_VALUE) {
            // Scale PID values since EEPROM values are unscaled
            thermalManager.temp_bed.pid.Kp = pid.Kp;
            thermalManager.temp_bed.pid.Ki = scalePID_i(pid.Ki);
            thermalManager.temp_bed.pid.Kd = scalePID_d(pid.Kd);
          }
        #endif
      }
```

You can test this by sending M304 P3000, M500, and then rebooting the board. This is present in both 1.1.x and 2.0.x branches.

ellensp, 2020-03-06:
So not a bug, but an undocumented feature. 
So should probably display a warning the console and LCD is there is one.

Alfrederson, 2020-03-06:
Or just remove the comparison with the dummy pid_value. Someone must have added it to fix a previous bug, but in doing so, created a trap.

thinkyhead, 2020-03-06:
We'll change `DUMMY_PID_VALUE` to `NAN`.

commit 913de024 (2020-03-06): Change PID dummy value to NAN

Fixes #17078
PR #17079 (2020-03-06): Change DUMMY_PID_VALUE to NAN
Fixes #17078

This PR changes the PID dummy value to NAN so that any PID values may be set to "3000" and still work.
closed 2020-03-06 by thinkyhead