MarlinFirmware/Marlin #17080: [BUG] M913 storing wrong inputs

AnimeEd, 2020-03-06:
### Bug Description

Sending M913 gcode via USB. Storing wrong inputs

### My Configurations

SKV Mini E3 running 2.0.4.4

### Steps to Reproduce

1. Send M913 X160
2. Send M913 to see stored value
3. Marlin returns 

```
>>> M913
SENDING:M913
X stealthChop max speed: 162
Y stealthChop max speed: 100
Z stealthChop max speed: 3
E stealthChop max speed: 30
```

**Expected behavior:** Expect X stealthChop max speed to be 160

**Actual behavior:** 162 speed is stored

#### Additional Information

Here are some other store values from different inputs

M913 X158
returns 159

M913 X100
returns 100

M913 X110
returns 111

M913 X200
returns 201

thinkyhead, 2020-03-06:
This appears to be normal behavior for TMC drivers. The values are set in the drivers and then read back, so the values you see are the ones the drivers have internally. @teemuatlut can correct me if I'm wrong.

closed 2020-03-06 by thinkyhead

AnimeEd, 2020-03-06:
I see.. so how come when it is set using the knob and display, it doesn't have this same effect?

thinkyhead, 2020-03-06:
In M913 you're setting the `pwm_thrs`. In the menu you're setting the `hybrid_thrs`.

teemuatlut, 2020-03-07:
The value is stored as a whole number.
So lets say `9/4 = 2.25 = 2`
And when you read the value: `2 * 4 = 8`

thinkyhead, 2020-03-08:
I'm not sure if it makes sense to have another g-code for setting `hybrid_thrs` since we have it in the menu. And I'm not sure if it makes sense to have a menu item for setting `hybrid_thrs` since we have it as a g-code. Usually we try to have both, but it comes down to what is best to expose for common usage.

swilkens, 2020-05-11:
Can we convert M913 to set `hybrid_thrs` and only expose that to the user, and set the stepper driver in the background?

Just to make the user-side less confusing.