MarlinFirmware/Marlin #17081: [BUG] (short description)

ManuelMcLure, 2020-03-07:
### Bug Description

Compile error on latest bugfix-2.0.x
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4301302/Configuration.zip)

### My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4301302/Configuration.zip)

### Steps to Reproduce
Download bugfix-2.0.x commit 3b141589797de1e1cb7071c272cae7d80c93e9bb
Using the included configuration files and ``default_envs = LPC1768`` attempt to compile in PlatformIO

**Expected behavior:** Compiled firmware

**Actual behavior:** The following errors:
```
In file included from d:\git\marlin\marlin\src\pins\pinsdebug.h:175,
                 from Marlin\src\gcode\config\M43.cpp:29:
d:\git\marlin\marlin\src\pins\pinsdebug.h: In function 'void report_pin_state_extended(LPC176x::pin_t, bool, bool, const char*)':
d:\git\marlin\marlin\src\hal\hal_lpc1768\pinsdebug.h:29:30: error: assignment of read-only parameter 'pin'
   29 | #define pwm_details(pin) pin = pin    // do nothing  // print PWM details
d:\git\marlin\marlin\src\pins\pinsdebug.h:250:46: note: in expansion of macro 'pwm_details'
  250 |         if (pin_is_protected(pin) && !ignore)
      |                                              ^
d:\git\marlin\marlin\src\hal\hal_lpc1768\pinsdebug.h:29:30: error: assignment of read-only parameter 'pin'
   29 | #define pwm_details(pin) pin = pin    // do nothing  // print PWM details
d:\git\marlin\marlin\src\pins\pinsdebug.h:298:11: note: in expansion of macro 'pwm_details'
  298 |       if (alt_pin_echo(pin)) {
      |           ^~~~~~~~~~~
Compiling .pio\build\LPC1768\src\src\gcode\control\M120_M121.cpp.o
*** [.pio\build\LPC1768\src\src\gcode\config\M43.cpp.o] Error 1
```

#### Additional Information
I'm pretty sure this was introduced by the change to using ``const pin_t`` in the parameter list of ``report_pin_state_extended()`` in commit ca4423ed2a8188bcd671ef59c25765e44e67fbcf.

thisiskeithb, 2020-03-07:
I'm seeing this as well. Temporarily disabling `PINS_DEBUGGING` will let you compile.

closed 2020-03-07 by thinkyhead

thisiskeithb, 2020-03-07:
@thinkyhead just pushed a commit to fix this: https://github.com/MarlinFirmware/Marlin/commit/e9f451bffece18ac90b7c34e3b06e3cdace57911.