MarlinFirmware/Marlin #17085: [BUG] Reprapdiscount Smart Controller shows garbage text after boot screen

mysteryjeans, 2020-03-07:
RRD smart controller shows garbage text after bootscreen, I have test with default configuration of 2.0.4.4 with only following changes as can be seen in picture. 

#define SDSUPPORT
#define SPEAKER
#define REPRAP_DISCOUNT_SMART_CONTROLLER

I have also tested with 1.1.9 (currently running with my setup), it doesn't even shows boot screen.

[image]

[image]

mysteryjeans, 2020-03-07:
This simple hello world example is working fine, so ruled out the hardware issue. On back of my LCD mentioned 2004A not just 2004.
```
#include <LiquidCrystal.h>
LiquidCrystal lcd(16, 17, 23, 25, 27, 29);

void setup() {
  // put your setup code here, to run once:
  lcd.begin(20, 4);
}

void loop() { 
  lcd.home();
  lcd.print("hello, world!");
  delay(500);
  lcd.clear();
  delay(500);
}

```

thinkyhead, 2020-03-08:
This is not a bug in the firmware. You have a bad connection of some kind.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-03-08 by thinkyhead

mysteryjeans, 2020-03-08:
Don't want to question you authority on this, but how come its a bad connection if hello world example running just fine? I will share screenshot later

thinkyhead, 2020-03-08:
Marlin does some things the simpler sketch does not, like running at higher speeds, going direct to port bits, and so on. The displays are therefore more sensitive to connection issues.

mysteryjeans, 2020-03-11:
You were right! extruder stepper motors was making clicking sound on printing, since the E0 was even pins and display pins are odd ones. I swapped the E0 with E1 and used E1 slot on the Ramp, clicking noise is gone while printing with RRD enabled in the config.

This leads me to think problem might be in the Ramps or Mega board itself.

```
#define LCD_PINS_D4 23 
#define LCD_PINS_D5 25 
#define LCD_PINS_D6 27
#define LCD_PINS_D7 29

#define E0_STEP_PIN        26
#define E0_DIR_PIN         28
#define E0_ENABLE_PIN      24
#ifndef E0_CS_PIN
  #define E0_CS_PIN        42
#endif
```

Printing while E0 was on default setup and SDSUPPORT disabled, display look like this. However swapping E0 with E1 display went back to grabled text.
[image]