MarlinFirmware/Marlin #17086: Fysetc AIO II only half heater PWM

Ksh-germany, 2020-03-07:
PID Tuning Shows only half pwm (127) so heater overshots 4°C (peek-to-peek 8°C)

i testet example config without changing anything, same effect.

sjasonsmith, 2020-03-07:
Please follow the instructions in the template and post copies of your configuration files.

After doing that, please explain how you are observing the issue, so that somebody else knows what to look for.

Ksh-germany, 2020-03-08:
As your wish, my config and some Picture from  the Problem

Pic 1 is at PID tuning
[image]
Pic 2 is after saved PID values and rebootet board
[image]

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4303243/Marlin.zip)

ellensp, 2020-03-08:
Please read https://reprap.org/forum/read.php?146,554826
"Marlin does not show the PWM (as one would expect!) from 0 - 255, but somehow only from 0-127 - i.e. the least significant bit is lost. So if you see "...B@:127" in pronterface, this means you have a full 100% duty cycle"

Ksh-germany, 2020-03-08:
hmmmm.... 5 Printer Show there 255 only this board Shows 127 and Need more time to heat up (i wanna Change an board)

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle