MarlinFirmware/Marlin #17089: SKR Pro V1.1 with BLTouch wont home.

xeddog, 2020-03-08:
### Bug Description

I am installing the SKR Pro in my Ender 5 Plus and have everything working except the BLTouch.  When homing the machine, X homes, then Y homes, then the carriage moves to the center of the bed.  The probe then deploys the pin, the machine completes the first probe, and then lowers the bed again getting ready for the second probe.  Probe deploys the pin again and the bed starts coming up.  When the pin touches the bed, the bed stops, and the red light in the probe comes on, but the probe action does not complete.  The terminal in Octoprint shows a serial communications error and disconnects.  The only indication on the display ( a real RRDFG 12864 display) just shows "STOPPED." on the bottom line.

### My Configurations

[Config_and_Pins.zip](https://github.com/MarlinFirmware/Marlin/files/4302739/Config_and_Pins.zip)

### Steps to Reproduce
After a failure, I have to 
1. Power cycle the printer
2. Connect with Octoprint and go the the terminal
3. Issue M502, M500 commands, then M280 P0 S60 (reset), M280 P0 S10 (deploy pin), and M280 P0 S90 (retract pin)
4. Home machine again.
Note:  I don't think any of the above commands can be skipped.

**Expected behavior:** [What you expect to happen]
Homing to complete
**Actual behavior:** [What actually happens]
Homing fails
#### Additional Information
Lots of things going on.  SKR PRO V1.1 with TMC2209 drivers for X, Y, and Z in SpreadCycle mode.  Also, since the SKR Pro has no EEPROM, I have added an external module, and for everything else it seems to be working perfectly.

thinkyhead, 2020-03-08:
First thing that can help is to turn off the option in OctoPrint to disconnect when it sees the word "Error." This will keep the printer connected.

---

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

xeddog, 2020-03-09:
Big oops on my part there.  I have three printers giving me a ration, and it is one of the others that disconnects.  This printer does not disconnect, but this shows up on the Octoprint terminal:

Recv: Error:!! STOP called because of BLTouch error - restart with M999
Recv: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
Recv: //action:notification STOPPED.
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: X:145.00 Y:183.00 Z:10.00 E:0.00 Count X:11600 Y:14640 Z:8000
Recv: ok P15 B3
Send: M113 S4
Recv: ok P15 B3

The rest of the information I think is good.  But the only way to "clear" the probe for another test is to power down, and back. Then the commands listed.  The M502/M500 must also be performed.  If I skip those two commands, after powering back up and running all of the M280 commands listed, the probe will not deploy the pin at all.  Period.

davidcgu, 2020-03-10:
Where you have connected the signal of the bltouch?  can you try to see what happens when is connected to Zmin?

xeddog, 2020-03-10:
> 
> 
> Where you have connected the signal of the bltouch? can you try to see what happens when is connected to Zmin?

I have tried Z-Min and also the PA2 pin on the "Probe" header pins using a variety of configurations.  The closest I can get is using the config I attached.  Z_MAX_PIN and Z_MAX_PROBE_PIN set to PG8 and using the Z-Min plug.

davidcgu, 2020-03-10:
I have it working wihtou problem...
[image]

xeddog, 2020-03-10:
I have spent a good part of the day today trying to get this working with bugfix-2.0.x, and have not been able to successfully get the BLTouch working.  I get the same results as with 2.0.4.4.  Most of the configurations I tried were Hail Mary's, but Mary just said Hail NO!!!

> 
> 
> I have it working wihtou problem...
> [image]

That is exactly how I have it wired now.  Still no joy.  Any chance you post your configuration files?  (Configuration.h, configuration.adv.h, and pins)

davidcgu, 2020-03-10:
try 2.0.1.....

xeddog, 2020-03-11:
> 
> 
> try 2.0.1.....

I just tried using Marlin 2.0.1 and the BLTouch works, and 2.0.4.4 and bugfix-2.0.x do not.  Still had an issue with UBL levelling, but that is another problem.

CarlitoGrey, 2020-03-11:
Hey. I’m on mobile so I can’t view your config but I have a bltouch on an sky 1.1 pro and tmc2208’s. 

The probe is plugged in to the dedicated probe pins and my confit is linked if you want to take a look and compare. 

  https://github.com/CarlitoGrey/Marlin Up to date as of 2.0.4.4.

xeddog, 2020-03-15:
I just tried using 2.0.5 and the problem remains, so back to 2.0.1 which works better, but has problems too.

GhostlyCrowd, 2020-04-18:
Im encountering this issue on a pull from today, 

As soon as it deploys the probe it retracts it deploys it another 2 times then errors

boelle, 2020-06-21:
@xeddog is this still an issue?

xeddog, 2020-06-21:
I had forgotten about this,  sorry.  This is no longer an issue for me, only because after two or three weeks trying to get it working, I took the BLTouch off and installed an EZABL Pro Mini.  Problem solved.  
I had also neglected to add one piece of information that turns out **may** have been very important.  I had moved the control box out of the frame and in so doing I had to extend or re-route all of the cables.  I made new cables for most of the other wires, but I didn't have the proper connectors and contacts to do the BLTouch cable properly and I had to make-shift one.  Continuity seemed good though as I poked/prodded/twisted/coiled/yanked etc., and never saw continuity break.  I had also posted this on a Facebook forum, and one of the responders mentioned that BLTouch was VERY susceptible to noise in the wiring, so maybe that was the issue here.

Thank you for following up, and I apologize for not closing this sooner

closed 2020-06-21 by xeddog