MarlinFirmware/Marlin #17090: [BUG] SKR 1.3 Pullups only work on 2 of 6 Endstop inputs

daviid5, 2020-03-08:
### Bug Description

I have enabled pullups for all endtsop inputs, but only on Y_MAX und Z_MAX I can measure them.
All other endstop inputs have 0V.
I'm not shure if this is a bug, a configuration error (the config is from a frined of mine, I only changed it to CoreXY und some other minor changes (I'm using X_MAX, Y_MAX, Z_MIN and not X_MIN, Y_MIN, Z_MIN/Z-Probe like he), or it's an hardware problem?

### My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4303063/Configuration.zip)

### Steps to Reproduce

Build a firmware with the config.
Flash it to the board.

**Expected behavior:** 
Pullups should be active on all endstop inputs.
If I disconnect all endstop switches, and call M119 it should like this:
11:09:51.420 : x_max: TRIGGERED
11:09:51.420 : y_max: TRIGGERED
11:09:51.420 : z_min: TRIGGERED

**Actual behavior:**
But if I call M119 it looks like this:
11:09:51.420 : x_max: open
11:09:51.420 : y_max: TRIGGERED
11:09:51.420 : z_min: open

Which makes sense because I cannot measure an high level on x_max and z_min port with an multimeter ...

Does anybody have an Idea?

ellensp, 2020-03-08:
The pullups on the SKR 1.3 are physical resistors on the board. 10K each. So enabling or disabling them in firmware has little effect.

daviid5, 2020-03-08:
@ellensp 
correct.
The firmware enables/disables the pulldown. This seems to happen, or the microcontroller is defect.
If I measure across the pullups I get the following (of course without power on the board):
R9 = 10k
R5 = 4k
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-SCH.pdf
Which seems to be an hardware defect in the microcontroller for those pins :(

ManuelMcLure, 2020-03-08:
You are not understanding. The Marlin pull up settings just enable or disable the internal pull ups on the processor. The SKR 1.3 has additional hard wired pull ups on the PCB which _can not be disabled by firmware_.
So if you are not measuring pull ups on the board then there is a hardware issue and nothing Marlin can do about this.

borland1, 2020-03-10:
LPC176x processor runs at 3.3V and I/O pins are not 5 volt tolerant, while the on board limit switch connectors are provided with 5V, Gnd, and signal terminals.  So resistors were added in cases when 5v signal limit sensors (three wire) are used.

ManuelMcLure, 2020-03-10:
The I/O pins on LPC176x _are_ 5 volt tolerant unless they are used as analog inputs. The resistors have nothing to do with 5V tolerance - they are pull-ups.

thinkyhead, 2020-03-11:
Any point in adding a sanity check to throw a `#warning` whenever pointless pullups are perfunctorily posited?

jafidod, 2020-03-29:
I've faced similar/same issue and with endstop settings at SKR 1.3 today and in my case just Z_MAX sensing pin wasn't affected. When I started digging into it I've found that XST, YST, ZST and E0S, E1S where shorted with the corresponding endstop sensing pins and causing troubles with the pull-ups. In case you can live without sensorless homing, it is worthy to don't use it.

daviid5, 2020-04-10:
Sorry for the long delay, and thank you for your suggestions :)
Well I found the problem was a wrong jumper configuration, really emberassing ...
Sorry for this stupid questions.

closed 2020-04-10 by daviid5