MarlinFirmware/Marlin #17091: [BUG] SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z

Lino77, 2020-03-08:
In file included from Marlin/src/HAL/HAL_LPC1768/../../core/../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/../../core/serial.h:24,
                 from Marlin/src/HAL/HAL_LPC1768/DebugMonitor.cpp:26:
Marlin/src/HAL/HAL_LPC1768/../../core/../inc/SanityCheck.h:2307:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
 2307 |     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
      |      ^~~~~
Compiling .pio/build/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_sdcard.cpp.o
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/HAL.cpp:25:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:2307:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
 2307 |     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
      |      ^~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/HAL_SPI.cpp:51:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:2307:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
 2307 |     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
      |      ^~~~~
Compiling .pio/build/LPC1769/src/src/HAL/HAL_LPC1768/timers.cpp.o
In file included from Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/../../sd/cardreader.h:24,
                 from Marlin/src/HAL/HAL_LPC1768/main.cpp:38:
Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/SanityCheck.h:2307:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
 2307 |     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
      |      ^~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/persistent_store_sdcard.cpp:29:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:2307:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
 2307 |     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
      |      ^~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/persistent_store_flash.cpp:44:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:2307:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
 2307 |     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
      |      ^~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:38,
                 from Marlin/src/HAL/HAL_LPC1768/timers.cpp:31:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:2307:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
 2307 |     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
      |      ^~~~~
Compiling .pio/build/LPC1769/src/src/HAL/HAL_LPC1768/u8g/LCD_I2C_routines.cpp.o
*** [.pio/build/LPC1769/src/src/HAL/HAL_LPC1768/DebugMonitor.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/HAL_LPC1768/HAL.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/HAL_LPC1768/HAL_SPI.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/HAL_LPC1768/main.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_sdcard.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/HAL_LPC1768/persistent_store_flash.cpp.o] Error 1
*** [.pio/build/LPC1769/src/src/HAL/HAL_LPC1768/timers.cpp.o] Error 1
================================================================================== [FAILED] Took 2.34 seconds ==================================================================================

Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
rambo                     IGNORED
FYSETC_F6_13              IGNORED
FYSETC_F6_14              IGNORED
sanguino644p              IGNORED
sanguino1284p             IGNORED
melzi                     IGNORED
melzi_optiboot            IGNORED
at90usb1286_cdc           IGNORED
at90usb1286_dfu           IGNORED
DUE                       IGNORED
DUE_USB                   IGNORED
DUE_debug                 IGNORED
LPC1768                   IGNORED
LPC1769                   FAILED    00:00:02.336
STM32F103RC               IGNORED
STM32F103RC_fysetc        IGNORED
STM32F103RC_btt           IGNORED
STM32F103RC_btt_USB       IGNORED
STM32F103RC_btt_512K      IGNORED
STM32F103RC_btt_512K_USB  IGNORED
STM32F103RE               IGNORED
STM32F103RE_btt           IGNORED
STM32F103RE_btt_USB       IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
STM32F103VE_GTM32         IGNORED
STM32F103VE_longer        IGNORED
mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
mks_robin                 IGNORED
mks_robin_pro             IGNORED
mks_robin_lite            IGNORED
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
chitu_f103                IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            IGNORED
rumba32_mks               IGNORED
include_tree              IGNORED
============================================================================ 1 failed, 0 succeeded in 00:00:02.336 ============================================================================
Der Terminalprozess wurde mit folgendem Exitcode beendet: 1

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4303079/config.zip)

ellensp, 2020-03-08:
Not a bug. A configuration issue.
You have #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN when you don't have a probe.

ellensp, 2020-03-08:
But I do think the error could be improved to reflect the true issue.

Patag, 2020-03-08:
Nope
Same kind of error here with Z2 on a dual Z configuration. Sanitycheck changed since 06 March and broke compilation (worked fine before).

ellensp, 2020-03-08:
@Patag you issue is unrelated to this one. 
With the above change this configuration compiles as expected

Patag, 2020-03-08:
OK. So i'll open a new issue report...

Lino77, 2020-03-08:
okay now my mistake is compiled

closed 2020-03-08 by Lino77