MarlinFirmware/Marlin #17092: [FR] Adjust Z probe offset based on temperatures

KaSroka, 2020-03-08:
### Description
I have noticed that my perfect Z probe offset varies with bed and nozzle temperature. It would be nice to allow for temperature compensation of this value.

Those are my measurements:

Bed temp | Nozzle temp | Perfect Z probe offset
-- | -- | --
50 | 200 | 1,075
70 | 240 | 1,025
110 | 240 | 1

From my measurements I've got those coefficients: 
Bed temp coefficient -0.00062
Nozzle temp coefficient 0.000938

### Feature Workflow
We may add new parameters to the configuration:
```
#define Z_PROBE_OFFSET_BED_TEMP_COEFF -0.00062
#define Z_PROBE_OFFSET_NOZZLE_TEMP_COEFF 0.000938
#define Z_PROBE_OFFSET_BED_TEMP_CALIB 110
#define Z_PROBE_OFFSET_NOZZLE_TEMP_CALIB 240
```

And adjust probe Z offset based on current temperatures:
`adjusted_z_offset = z_offset + (Z_PROBE_OFFSET_NOZZLE_TEMP_CALIB - current_nozzle_temp) * Z_PROBE_OFFSET_NOZZLE_TEMP_COEFF + (Z_PROBE_OFFSET_BED_TEMP_CALIB - current_bed_temp) * Z_PROBE_OFFSET_BED_TEMP_COEFF`

swilkens, 2020-03-08:
The default for the coefficient would probably have to be "1" or the entire feature may have to be off by default, just to ensure users with beds that are less / non-reactive in geometry to heat that level the bed at room temp don't suddenly have a surprise deviation at printing temperature.

I like the FR though, it makes sense to account for bed warping due to temperature increase - especially on the thinner beds. It might even be worked into UBL @Roxy-3D ?

Though I wonder if these CTE geometry changes are within the error range of typical probes, in that case it's problematic.

KaSroka, 2020-03-08:
Those measurements I've got are consistent (tested on at least 4 prints each). I'm using BL-touch from Ali. And I've verified that first layer is always 0.1mm (+/- 0.01mm)

The default should be 0 but I think that disabling this feature by default is a better option.

davidcgu, 2020-03-09:
try a pinda, it's already implemented since 2.0,.3

closed 2021-11-02 by thisiskeithb