MarlinFirmware/Marlin #17094: [BUG] Compilation broken since March 06

Patag, 2020-03-08:
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:38,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:2088:4: error: #error "An SPI driven TMC driver on Z2 requires Z2_CS_PIN."
   #error "An SPI driven TMC driver on Z2 requires Z2_CS_PIN."

Was compiling fine before, so no need of my config files

ellensp, 2020-03-08:
no config files, cannot help you without them.

tureka, 2020-03-08:
I have the same problem. Stable branch is compiling without problems and bugfix shows this error. Freshly downloaded both and changed only settings for steppers and tmc software spi. This is shown with dual Z steppers option enabled with a stepper driver connected to E1. I tried to compile with TMC5160 and TMC2209.

oliver-eifler, 2020-03-08:
Same error with BTT SKR Pro/tmc2130-spi/dual-Z and current bugfix-2.0.x ( https://github.com/MarlinFirmware/Marlin/tree/2a71e4f336e66d93983ca15714137085090a73b8)
it compiles fine when redefining needed pins in configuration_adv.h
`#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 1
....
#endif
#undef E1_STEP_PIN        
#undef E1_DIR_PIN         
#undef E1_ENABLE_PIN      
#ifdef E1_CS_PIN
  #undef E1_CS_PIN        
#endif
#define Z2_STEP_PIN        PD15
#define Z2_DIR_PIN         PE7
#define Z2_ENABLE_PIN      PA3
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN        PG15
#endif
`
current configuration.h/configuration_adv.h/platform.ini - not for a real printer yet ;)
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4303461/configs.zip)

ellensp, 2020-03-08:
you do not #define Z2_DRIVER_TYPE TMC2130 when you are really using E1
you set #define E1_DRIVER_TYPE TMC2130 
The pins.h file notices you have Z > 1  and defines all the Z2 pins from the E1 pins 
You only set #define Z2_DRIVER_TYPE TMC2130 if you have actual Z2 pins defined in your boards pins.h file.

Patag, 2020-03-08:
@ellensp your response is really boring... 
anyway 
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/4303542/conf.zip)

GMagician, 2020-03-08:
@patag I think you should have a little more respect of people who want to help you, considering that attaching configuration is required when posting an issue

Patag, 2020-03-08:
@GMagician . You're right for sure. But please don't take me for a noob. By the way , did you test the last merge ?

oliver-eifler, 2020-03-08:
defining E1_DRIVER_TYPE TMC2130 instead of Z2_DRIVER_TYPE TMC2130 compiles but its very unlogical as any other configurastion is base on axis name  and it has worked in the past
same error/workaround for X_DUAL_STEPPER_DRIVERS, Y_DUAL_STEPPER_DRIVERS and NUM_Z_STEPPER_DRIVERS 2/3/4

ellensp, 2020-03-08:
In previous versions it didn't check that a  Z2_CS_PIN pin existed when you set #define Z2_DRIVER_TYPE. Now it correctly does. 

Z2 in the past would have been configured as a A4988 (the default E1 settings) 
  
It is very logical.
The board pin definition only have definitions for the stepper drivers that are on the board.
It redefines unused E steppers for other axis if you use such things.

oliver-eifler, 2020-03-08:
I mean its unlogical that Z2_DRIVER_TYPE TMC2130 doesn't work but #if AXIS_IS_TMC(Z2) works
neverthanless, from now on i'll redifine the required pins

tureka, 2020-03-08:
Then I simply set `#define Z2_DRIVER_TYPE TMC2130` in Configuration.h
Then in Configuration_adv.h

`#define NUM_Z_STEPPER_DRIVERS 2`
`.....`
`#if AXIS_IS_TMC(Z2)
    #define Z2_CURRENT      760 // *LT 800
    #define Z2_CURRENT_HOME Z2_CURRENT
    #define Z2_MICROSTEPS    16
    #define Z2_RSENSE         0.075 // *LT 0.11
    #define Z2_CHAIN_POS     -1
  #endif`
`.....`
`#define Z2_CS_PIN         P1_01` assign pin to E1 CS pin on board.

Is it correct thinking??? 

Otherwise if `#define E1_DRIVER_TYPE TMC2130` and all other settings for Z2 axis I am not able to set currents with M122 or M914 for StallGuard sensivity for this stepper.

ellensp, 2020-03-08:
@tureka There does look to be an issue here...  the gcodes expect Z2_DRIVER_TYPE while the config seems to expect E1_DRIVER_TYPE

thinkyhead, 2020-03-08:
Originally, assigning driver types to axes was meant to apply based on the stepper driver positions on the board. So your E1 would be defined and you wouldn't need to define a Z2 unless the board actually had a space designated specifically for a Z2 plug. But some of the order of inclusion and dependency related most especially to Trinamic drivers kind of threw this off. The automatic assignment of X2, Y2, Z2, etc. to the next unused En plug is in `pins.h` which is only included after the two `Configuration` files. So the compromise was made to assign driver types directly but leave the axis reassignment in place for pins.

So at least according to my understanding, `Z2_DRIVER_TYPE` should apply for the Z2 axis, and the `E1_DRIVER_TYPE` should only apply to a real E1 extruder.

thinkyhead, 2020-03-08:
@tureka — You should be applying your settings on the "Z2" axis in in `Configuration_adv.h` because Marlin has no idea at that point that there is any axis reassignment in front of it, and the intermediate stuff that comes in-between `Configuration.h` and `Configuration_adv.h` — which includes all our Trinamic driver macros — also has no idea about our axis reassignment voodoo.

So, as described above, we have two paradigms. In the PINS paradigm you define based on the names silkscreened on the board. And in the DRIVERS paradigm we define based on the axis function within Marlin.

thinkyhead, 2020-03-08:
I note that yes you must currently make sure your `E1` driver type matches up with your `Z2` otherwise Marlin nukes the `E1_CS_PIN`.

GMagician, 2020-03-08:
> You're right for sure. But please don't take me for a noob. By the way , did you test the last merge ?

@Patag not meant to take you as noob.
I can't test this board 'cause I don't have it

ellensp, 2020-03-08:
@thinkyhead With both #define E1_DRIVER_TYPE TMC2130 and #define X2_DRIVER_TYPE TMC2130   M906 fails to show either E1 or X2

thinkyhead, 2020-03-08:
The patch I just merged will fix the general issue where the `E1_CS_PIN` was getting eaten.

closed 2020-03-08 by thinkyhead