MarlinFirmware/Marlin #17095: Z_HOMING_HEIGHT improvment

1Alfa1, 2020-03-08:
### Description
Working with MPCNC and MARLIN 414D Bugfix 2.0.x for dual endstops flavor from V1Engineering.

After much work to get Z_HOMING_HEIGHT working through GITHUB and other research:
notes on my parameters:
changed Z_MAX_ENDSTOP_INVERTING to true (THANK YOU BLAIRQ!)
changed in G28.cpp
//destination.z = z_homing_height;
      destination.z = TEST(axis_known_position, Z_AXIS) ? z_homing_height : current_position.z + z_homing_height;
Didn't go back to original "destination.z = z_homing_height"  due to what I read about this.
*Side note - see no difference with the below, (What was this about?) 
destination.z = z_homing_height + (TEST(axis_known_position, Z_AXIS) ? 0.0f : current_position.z);

I am using Z_SAFE_HOMING and have my Z touch sensor on other side of bed, also using probe feature with a servo idea I developed to avoid using alligator clip on endmill, to touch top of spindle armature because case is no conductivity / not reliable. I have other changes as well but I don't think any are specific to the need/benefit for this improvement.

Change G28.cpp to include z in IF statement on whether to do Raise Z before homing.
### Feature Workflow

1. Home All (unknown or with Z<Z_HOMIING_HEIGHT) and Z raise before home works
2. Move X and/or Y around and then Z lower than Z_HOMING_HEIGHT
3. Use custom command "HOME Z AXIS" and it could crash and endmill (or nozzle).
4. Change line in G28.cpp from:
if (z_homing_height && (doX || doY)) { 
to
 if (z_homing_height && (doX || doY || doZ)) {
5. Now Z is raised even if doing a Z only home.

#### Additional Information
I apologize in advance if I am not following protocol for issues, I am new to ALL of this. Just trying to contribute.

thinkyhead, 2020-03-08:
> I apologize in advance if I am not following protocol for issues

Were the instructions in the issue template unclear? Your post is very hard to follow. If you would include your configurations as explicitly requested then we won't have to harass you with a hundred questions.

thinkyhead, 2020-03-08:
So, here's my first of a hundred questions while we wait for your configurations.

Are you using `Z_SAFE_HOMING` and is your problem due to the fact that XY needs to do a sideways move before the Z axis can home? Or do you have some kind of probe that won't work if it starts out too low to the bed?

The Z raise is set up to do a raise over the current position when the current Z position is "unknown" and to do a raise to the set height when the Z position is known.

1Alfa1, 2020-03-09:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4304079/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4304080/Configuration_adv.txt)
the video just shows how I am using the Probe option it does not show the "undesired behaviour" that I speak of in step 3 of feature workflow.

1Alfa1, 2020-03-09:
Video not working.... here is picture
[image]

1Alfa1, 2020-03-09:
Let me try to word this better. I know I added to much irrelevant info but I had so much trouble getting other things to work first, I thought it would be helpful. If I home and then move the head around manually and lower the head below the 0 position.. i.e below the touch sensor which is this (https://www.amazon.com/Z-Axis-Router-Touch-Setting-Milling/dp/B07GT3CHWZ/ref=sr_1_2?keywords=touch+plate+for+CNC+machine&qid=1583764743&sr=8-2). Then if I do a "Z HOME" (i.e. not re home x or y). It will crash. BUT if you add || doZ to the right place in the G28.cpp code... it appears to fix that problem. (Not sure if it would cause another issue).
[image] Thank you to all the creators Marlin is awesome!

boelle, 2020-06-20:
@1Alfa1 is this still an issue? or is this purely support questions?
@thinkyhead

closed 2020-06-23 by boelle