MarlinFirmware/Marlin #17096: [BUG] Very high speed to Z SAVE position

kaeltaz, 2020-03-08:
Hi
when I turn on auto home, X and Y axis is homing on normal speed then when hotend moves to z save position at very high speed (probably at maximum feedrate) 
How to slow down this movement?

ps. sorry for my bad English :)

kaeltaz, 2020-03-09:
I use Marlin 2.0.4.4  
how to solve my problem?

psavva, 2020-03-09:
Please pay your config

kaeltaz, 2020-03-09:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4309095/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/4309097/Configuration_adv.h.txt)

kaeltaz, 2020-03-11:
https://youtu.be/4enE0p8TQyQ 

Video

psavva, 2020-03-11:
One too many zeros here...
#define XY_PROBE_SPEED 120000

Vertabreak, 2020-03-11:
good lord that's fast i cant do more then 2400 with out really bad vibration.

closed 2020-03-11 by thinkyhead