MarlinFirmware/Marlin #17098: [BUG] "TMC connection error" with TMC2209 drivers

Lupek1, 2020-03-08:
Hello,
I have home made 3D printer, Ramps board with latest 2.0.4.4 Marlin release and Bigtreetech TMC2209 drivers on X and Y axis. I am able to move axis, but UART connection don't work. "TMC connection error" is shown on LCD all the time. 

I checked all connections multiple times, ask for support on FB page, seeking for solution on internet but infortunately, after one month, I still didn't manage to solve it. So I am asking you for some help.

Regards,
Luka
[TMC_issue.zip](https://github.com/MarlinFirmware/Marlin/files/4304007/TMC_issue.zip)

ellensp, 2020-03-09:
For support try the https://reprap.org/forum/index.php forums.
And you will need to provide much more information. Like brand/supplier of TMC2209 drivers. If you purchased them as UART enabled or you modified them so they work with UART. and how you have it wired it on the RAMPS, what pins you used. etc.

davidcgu, 2020-03-09:
[image]

Lupek1, 2020-03-09:
I use TMC drivers from Bigtreetech: 
https://www.aliexpress.com/item/33025720969.html?af=701906

they allready have UART connection to pin 4 by default. But they connect it using resistor, not weld. 
I keep resistor and make additional connection to pin 5 with weld. But TMC connection error is still there.
[image]

I use default pins. I believe driver is connected to right pins because as soon as I enable TMC, voltage on those pins go from 0V to 5V.
[image]
[image]

Function M122 gives me:
[image]

davidcgu, 2020-03-09:
Who has te issue? I don't know why you posted instructions, mine are working.... post a picture of the bottom of your drivers...

Lupek1, 2020-03-09:
Sure. Here is photo of my driver.

[image]

davidcgu, 2020-03-09:
The soldering joint is not correct, look at the picture I shared previosuly, should be like the one on the left side.

Regards

davidcgu, 2020-03-09:
What the fk!!!!!!!!  why you cutted all those pins?

Lupek1, 2020-03-09:
See original driver from another perspective. You can see that connection as you shown allready exist. It is made by factory. For connection they used small resistor around 1ohm. 

[image]

Ramps board and need external connection for UART. All pins pins I cut on bottom are placed on top side. But as far as I know, just one of them is needed  for UART connection.

[image]

Regards, 
Luka

ManuelMcLure, 2020-03-09:
The "resistor" is actually a 0-ohm jumper, so it acts just like a solder joint. It's easier for the automated equipment used to create these boards to place a component than make a solder bridge.

I just measured the resistance on all three of my BTT2209s and they were all around 0.10 ohm which is in the realm of measurement noise.

davidcgu, 2020-03-09:
why you don't consider to upgrade the board? you can get nice 32 bit boards by around 15€

Lupek1, 2020-03-09:
Davidcgu: In first stage, I would like to understand root cause. It makes no sense for me to purchase new board, altrough it is cheap if I don't know where the root cause is.

Do anyone have any experiances with UART protocol? What are frequencies? How often driver and arduino communicate with each other? I meassure TX and RX signals with 20MHz analog oscilloscope, but I see no signal. Just 5V constant voltage. Unfortunately I don't have enough knowledge to know either signals are too fast to see them with analog oscilostope either something is wrong with firmware.

ManuelMcLure, 2020-03-09:
Do you have any micro stepping jumpers installed under the drivers? For the 2209s those are interpreted as the "address" of the 2209 and if that address does not match the setting of (for example) ``X_SLAVE_ADDRESS`` in ``Configuration_adv.h`` then the communication will not work.
All slave addresses in ``Configuration_adv.h`` default to 0 which means no jumpers under the drivers.
Also, communication with the TMC drivers will _not_ work if the RAMPS does not have full 12V power. If it is being powered via USB the drivers will not be completely powered up and the communication will fail.

ManuelMcLure, 2020-03-09:
Also, please come on to the Marlin Discord server for support. This bug tracker is not meant for support issues.

Lupek1, 2020-03-09:
ManuelMcLure: I removed all 3 jumpers below driver for X and Y axis, where TMC drivers are used. For Z axis and two extruders I keep jumpers because  A4988 are used. Drivers are supplied with 12V from integrated PC power supply. X_SLAVE_ADDRESS are set to 0.

OK. I will open new topic on Marlin discord server. See you there. Thanks for support.

Regards, 
Luka

closed 2020-03-09 by Lupek1