MarlinFirmware/Marlin #17099: [BUG] Pausing (randomly) During Print -

Aetoc, 2020-03-09:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-03-09:
You don't have `FILAMENT_RUNOUT_SENSOR`, nor `ADVANCED_PAUSE_FEATURE` enabled, so perhaps it's your slicer.

Can you zip up & post some sliced .gcode that exhibits this behavior?

Also, it wouldn't hurt to update to the latest `bugfix-2.0.x` since it's updated daily.

psavva, 2020-03-09:
Try with a different slicer and see if you can reproduce the issue.

Aetoc, 2020-03-09:
[CFFFP_Body.zip](https://github.com/MarlinFirmware/Marlin/files/4309102/CFFFP_Body.zip)

This is the model it started happening with. It seems that it was only doing it during the printing of the raft. Eventually I went to bed and it continued overnight with no issues. I can't say 100% that it stopped pausing once the raft was completed, but it seems like it. 

I'm now printing an 8hr long print, and it has already paused once on the raft, and it's been 15 mins. I'll update if it continues. For the print following this one, I'll try a different slicer and see if that changes anything.

Sewyboowy, 2020-03-11:
Interesting, I am having a similar issue although I'm not using a raft. Let us know how it goes

thinkyhead, 2020-03-11:
Be sure to test the latest [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) so we're all on the same page.

tlhintoq, 2020-04-22:
Stop printing by USB.  Slice and save to SD card.
If it still happens there it might be an issue.  Otherwise its the USB or your computer.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle