MarlinFirmware/Marlin #17100: G29A successfully enabled UBL, but M420 S1 says mesh is invalid.

RFBomb, 2020-03-09:
### Bug Description

Unsure if this is actually a  bug or not, but my printer when performing "G29 P1" only hits 57/100 points then says it is done. 

### My Configurations
I am using BL-Touch on an Ender 3 with a SKR1.3 board. The probe, is attached to Z_Max Pin (per the SKR default probe pin). I am also using a microswitch on Z_Min. Here are my configured offsets:
M206 X0.00 Y0.00 Z-0.97
M851 X-44 Y-14 Z-0.7

[MyConfig.zip](https://github.com/MarlinFirmware/Marlin/files/4304628/MyConfig.zip)

### Steps to Reproduce

1. G28
2. G29 P1 -> it will probe 57/100 locations
3. M500 
4.1.  G29 
4.2 M420 S1 
 
[PronterFace_Console.txt](https://github.com/MarlinFirmware/Marlin/files/4304635/PronterFace_Console.txt)
attached above is the text i copied from the Pronterface console with everything I did. 

**Expected behavior:** [What you expect to happen]
According to the MarlinFW website, G29 A looks like it is just a shortcut for "M420 S1", so I would expect them to have the same output. 

**Actual behavior:** [What actually happens]

G29 A says the UBL becomes active, but M420 S1 won't activate the mesh because its invalid

#### Additional Information

Excerpt from the Pronterface Console
```
 M420 S1 L0
SENDING:M420 S1 L0
Mesh loaded from slot 0
Bed Topography Report:
    (  1,234)                                                              (234,234)
        0       1       2       3       4       5       6       7       8       9
 9 |   .       .       .       .       .       .       .       .       .       .
   |
 8 |   .     +2.090  +2.142  +2.252  +2.340  +2.410  +2.505  +2.555    .       .
   |
 7 |   .     +2.120  +2.177  +2.277  +2.357  +2.455  +2.530  +2.597    .       .
   |
 6 |   .     +2.112  +2.177  +2.247  +2.347  +2.440  +2.515  +2.570    .       .
   |
 5 |   .     +2.067  +2.145  +2.240  +2.337  +2.452  +2.552  +2.610    .       .
   |
 4 |   .     +2.097  +2.165  +2.260  +2.380  +2.482  +2.595  +2.640    .       .
   |
 3 |   .     +2.122  +2.197  +2.313  +2.412  +2.517  +2.625  +2.672    .       .
   |
 2 |   .     +2.140  +2.217  +2.330  +2.427  +2.537  +2.645  +2.705    .       .
   |
 1 |   .     +2.152  +2.235  +2.337  +2.427  +2.545  +2.650  +2.702    .       .
   |
 0 |   .       .       .       .       .       .       .       .       .       .
        0       1       2       3       4       5       6       7       8       9
    (  1,  1)                                                              (234,  1)
Mesh is invalid
Storage slot: 0
echo:Invalid mesh.
Error:Failed to enable Bed Leveling
[ERROR] Error:Failed to enable Bed Leveling

echo:Bed Leveling OFF
echo:Fade Height OFF
>>> M420 S1
SENDING:M420 S1
echo:Invalid mesh.
Error:Failed to enable Bed Leveling
[ERROR] Error:Failed to enable Bed Leveling

echo:Bed Leveling OFF
echo:Fade Height OFF
>>> G29 A
SENDING:G29 A
Unified Bed Leveling System v1.01 active
```

RFBomb, 2020-03-09:
Do I just have to use G29 P2 to fill in the other locations?

hliebscher, 2020-03-09:
I use G29 P3     why stops by 57 of 100 positions?   why you use a second microswitch ?

RFBomb, 2020-03-09:
I’ll try out the P3 argument and see how it goes, rather than the P2
argument which will require me to manually adjust the mesh. Thanks for that
tip.

As for the 57/100, I have no idea why it did that. When I executed G29 P2,
it brought the nozzle to locations that are perfectly reachable by the
probe (pretty much the whole border). So I’m not sure why G29 P1 skipped
all of those points.

Lastly, the use of the 2nd micro switch is mostly fit debugging right now,
I may disable it later on. But since I have an ender 3, I’m just using the
stick Z-0 switch for referencing the printer. This allows me to ensure it
should be able to test print with leveling disabled while I try to try to
figure out the BL-Touch. Essentially I’d rather work from a single known
good point than a floating point. (This stems from my job dealing with
Cartesian robots)

davidcgu, 2020-03-09:
G29 S1 to save the mesh on slot one.....

RFBomb, 2020-03-09:
When I was using Pronterface, the feedback from the printer said the mesh
was saved to slot 0.

The output I put into the description is from ‘M420 S1 L0’, (Load and
activate Mesh Slot 0). The topography report is from slot 0.

Why would saving it to a different slot make any difference?

RFBomb, 2020-03-09:
Also, I believe I may have a reason the border is missing: the default
setting in marlin is a probe offset of 10. I’ll try setting that to 0 since
I should be able to hit 0,0. (I’ll just have to provide a Y offset to 0,0
is actually with the nozzle on the bed)

davidcgu, 2020-03-09:
I used M420 for bilinear not for UBL.

As told... enable UBL is G29 A   store mesh is G29 S1  and load mesh on pos 1 is G29 L1.....

MAybe will be good to take a look on the docu:   https://marlinfw.org/docs/gcode/G029-ubl.html

Regards

RFBomb, 2020-03-09:
@davidcgu, I appreciate you trying to help, but I think your mistaken. From the page you linked:
In the first paragraph of the 'Notes' section:

> G28 disables bed leveling. Follow with M420 S to turn leveling on, or use RESTORE_LEVELING_AFTER_G28 to automatically keep leveling on after G28.

and the parameter section:

> [A<bool>] | Activate Unified Bed Leveling (i.e., M420 S1)

And finally from the M420 Page: https://marlinfw.org/docs/gcode/M420.html

> This command has been extended with L to load a mesh for Unified Bed Leveling.

So basically, according to all that, 'G29 L# A' and 'M420 S1 L#' should both load whatever the slot # is specified, and I would expect them to have the same result: Either activating or failure to activate. 

Even more so per this page: https://marlinfw.org/docs/features/unified_bed_leveling.html
> G29 A or M420 S1 | Activate the Z compensation bed leveling.

I hope that clears up your confusion on G29/M420. That said, I am finding it troubling that they are giving different results for what, according to the documentation, should be the same action. 

Once I am able to, I will humor you and save it to slot 1 instead of slot 0. Though I do not expect any difference in results, because slot 0 is where the firmware told me it saved to originally, and also was able to load from and provide a topography report from that slot. Plus, the only documentation I see with a special case is slot '-1', which just spits out raw data without saving it. I see no mention of slot 0, so I assume it is free to use (considering that's where the firmware told me it saved it).

davidcgu, 2020-03-09:
This works like this for linear not for UBL, but man this is a stupid discusion, I use UBL since long ago and is not me who has problems.

Good luck

RFBomb, 2020-03-09:
@davidcgu Like I said, I appreciate you trying to help. But the documentation explicitly states that it works with UBL on multiple pages. The issue report is that one says 'success' and the other says 'failure' despite supposedly being supported.

thinkyhead, 2020-03-11:
> According to the documentation, these two commands seem to be identical, so I'm confused as to why the results are success vs failure.

This may be a choice by design.

- `G29 A` simply activates without any fanfare and doesn't check for a valid mesh.
- `M420 S1` will only activate if the mesh is valid.

Also `G29 A` is not designed to be combined with `G29 L` since the `A` is apparently applied ahead of the `L`. However `M420` does permit load and activate in combination, since it loads first, then checks for a valid mesh, and then activates… in that order.

If you aren't working with U.B.L. specifically to refine and develop it, then it's best to stick with the more general `M420 S1`.

RFBomb, 2020-03-11:
So to clarify, I may have mentioned it but never actually used G29 A L, I
just reference it with shorthand for the discussion. I’ve only used G29 A,
never L. I’ll have to test what attempting to load that mesh with ‘G29 L0’
returns as a result.

If what your saying is true about activating without checking, does it
actually take effect?  (Related question: if I set an probe offset of 3mm,
will it raise print head up 3mm? If so, I can test like that to see if the
mesh became active in the plotted points.)

I still have to figure out why my initial mesh was reported as bad though
by M420.  I’m thinking it may be just because it’s missing points, but Im
not sure.

>> G29 A is not designed to be combined with G29 L since the A is a  and
develop it, then it's best to stick with the more general M420 S1.

Are you suggesting that due to the fact it validates the mesh?

Follow up: If the mesh is invalid, does that mean it just resorts to using
the standard 0 plane (as set by G28)?

thinkyhead, 2020-03-11:
> Are you suggesting that due to the fact it validates the mesh?

Yes. It will prevent activating leveling with an incomplete mesh.

> Follow up: If the mesh is invalid, does that mean it just resorts to using

It simply doesn't activate leveling.

RFBomb, 2020-03-12:
So does G29 A then activate leveling, even if the mesh is incomplete /
invalid?

Roxy-3D, 2020-03-12:
It is not necessary to have a fully populated (and valid) mesh to use UBL.   However, if you don't have a fully populated mesh...   You should be very careful to not print in the invalid mesh regions of the bed.    This was done deliberately because it was thought during the design and coding that people would not want to spend all the time necessary to populate the mesh.

But the amount of time it takes to generate the full mesh, and the ability to save the mesh to EEPROM pretty much made that concern a non-issue.

You should fully populate the mesh and then fine tune it with several iterations of G26 and G29 P4.   But you are in control.  You can do what ever makes sense for you and your printer.

> So does G29 A then activate leveling, even if the mesh is incomplete /
invalid?

YES!!!!

RFBomb, 2020-03-12:
Thank you for the clarification (finally)!
This should be noted on the marlinFW website to make it explicitly clear
how the interaction works, and that M420 will not work without a complete
and valid mesh, while G29 A works within the probed area.

If it were explained in clearer detail on the Marlin website, this entire
topic wouldn’t have been made.

Roxy-3D, 2020-03-12:
> This should be noted on the marlinFW website to make it explicitly clear
how the interaction works, and that M420 will not work without a complete
and valid mesh, while G29 A works within the probed area.

Ah...   I'll merge any Pull Request with better detail on the Bed Leveling documentation.   Please point out paragraphs or whole sections that should change!   It the documentation could be clearer...   That is good, but somebody has to identify the areas that should change and what the change should be....

RFBomb, 2020-03-12:
I’ll look over the pages again and make notes about it, then come back to
this post with another reply in a few days.

Roxy-3D, 2020-03-12:
Thanks!   Seriously, it will help new users if the documentation actually matched what the code is doing.   And that is across a whole bunch of different topics.    The documentation has not been getting any where near the same number of commits the code side has been getting.

The documentation side needs to catch up.    But somebody has to do that.

RFBomb, 2020-03-12:
@Roxy-3D  -- Before I publish my suggestions, I'll be asking some clarifying questions here. 
How does "RESTORE_LEVELING_AFTER_G28" work?
 - Does it simply issue a 'G29 A' / 'M420 S1' command at completion of G28? (If so which one?)
 - If neither of the above occurs, does it simply activate it (like G29 A) or does it validate it like M420?

Roxy-3D, 2020-03-13:
Actually...   It doesn't do either of those things.    The G28 code saves the state of the bed leveling activation and then turns bed leveling off.   Where ever you are probing to home the machine, you don't want any corrections to be performed.

Then...  at the end of the G28, if that option is enabled, it just blindly restores the state of the bed leveling activation.

But if you want to think about it at a higher level...   You could just assume it does an automatic G29 A (in the UBL case) at the end of the G28 homing operation.

RFBomb, 2020-03-13:
I use code like that all the time in my VBA projects lol. Thanks for the info. 
Also, did the format of the MarlinFW.org site change, or am I missing something? It seems like its organized differently than it was last week when I opened the issue. For example, I am having a hard time reaching the different versions of G29 and getting to M420 because they just aren't appearing in the sidebar at all for me. (using the hard-links to the web pages still works properly).  For reference, I'm using firefox. 
Its like the left-hand sidebar changes drastically, as if it was 'over-organized' and everything is buried/missing.  Here is all I see when viewing the G-Code index now on the side-bar (whereas previously wasn't it a list of all commands?)

```

    G0-G1: Linear Move
    G2-G3: Arc or Circle Move
    G17-G18: CNC Workspace Planes
    G38.2-G38.3: Probe target
    G54-G55: Workspace Coordinate System
    M0-M1: Unconditional stop
    M18-M84: Disable steppers
    M43 T: Toggle Pins

```

Roxy-3D, 2020-03-13:
Yes, things are moving around.    But I don't even pay attention to that.   I use Astro-Grep and just search for what ever function I want to mess with.    And then just double click on it to pull it into an editor to look at or change things.     I don't really keep track of where things are located in the hierarchy.

RFBomb, 2020-03-13:
I'm in process of writing a new file for G029. All the links that spawn from G29 are broken due to the file not existing. It will simply be a page to point to the various G29_ variants.   I'll have to look into AstroGrep and check it out.

RFBomb, 2020-03-13:
@Roxy-3D  OK, so I created several pull requests with my proposed changes, if you would like to review them/have another admin do so. 

As of right now, I'm only aware of M420 requiring a complete dataset in the mesh, I have no idea if there is anything else that will cause it to determine an 'invalid' mesh. If you know of anything, I left that modified section for 'validation requirements' formatted in such a way that it should be easy to add in all the things that it is checkign for validation. (For example, I'm assuming that it is checking within a reasonable tolerance so you dont have +/- 0.2 then all of a sudden +10. but as I am unsure if it does that I didn't write it down). 

also, on the G29_UBL page, i noticed this oddity:

> It's highly recommended to enable EEPROM. With EEPROM storage enabled, UBL is limited to
> 3-Point (G29 P0 T) and Grid (G29 P0 G) Leveling."
 
Should this say  `"With EEProm storage DISABLED, ....: ` ? It doesn't seem correct that UBL would be limited to those 2 modes with it saving to the EEPROM ENABLED, but I'm not a firmware guy. I left that with the original wording, but if it is a mistake, correcting it prior to merging would be appreciated.

closed 2020-03-13 by RFBomb

Roxy-3D, 2020-03-24:
> Should this say "With EEProm storage DISABLED, ....: ? It doesn't seem correct that UBL would be limited to those 2 modes with it saving to the EEPROM ENABLED, but I'm not a firmware guy. I left [...]

Can you provide a link?    In the UBL documentation I see this: (at https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md)

Enable these options:

AUTO_BED_LEVELING_UBL. Leave the others commented out.
G26_MESH_VALIDATION to enable the G26 test print utility.
EEPROM_SETTINGS. EEPROM support is required.
One of the LCD options. An LCD with a rotary encoder is strongly recommended.

thinkyhead, 2020-03-24:
> "It doesn't seem correct that UBL would be limited to those 2 modes"

I've already changed the documentation in response to that comment.

stableconnection, 2020-04-29:
Since this seems to be the issue closest to mine i will just ask here:
1. did removing the default 10mm like stated in the first(?) comment solve the incomplete mesh probing? if no, any ideas why it would stop at ~half?
2. why is the head moving in the wrong direction with G29 to get to center? (nozzle offset negated for whatever reason, cant find the cause of this but my nozzle offset is correct)
Also, what could be the reason for my setup not to take the G29 standalone command but only in combination with e.g. P1

sorry to bother y'all in a closed issue but this is kinda pressing and the only thread somewhat similar to my problem.
and thanks for clarifying the G29A and M420 thing, helps a lot.

RFBomb, 2020-04-29:
`Also, what could be the reason for my setup not to take the G29 standalone command but only in combination with e.g. P1`

G29 on its own does nothing without a followup parameter telling it what to do. Refer to the various options called out in the marlin documentation (such as "P1" /  "T" / etc).  This is normal behavior, as G29 is a bunch of different actions grouped into a single command because they are all related. The additional parameters you pass it determine which action is taken. 

`why is the head moving in the wrong direction with G29 to get to center`
What do you mean? If you have Z_Safe_Homing enabled, it will default to moving XY to the repsective (BedSize/2 - Probe Offset) position. So for a bed of 100/100, and probe offset of 20/10, it will move to (50 -20 ) / (50 -10) --->  X30/Y40.  If this is not what you are seeing, please clarify what you are in fact seeing. 

`did removing the default 10mm like stated in the first(?) comment solve the incomplete mesh probing? if no, any ideas why it would stop at ~half?`
Sort of.  If Think of it like this:
This is with the stock 10mm border. probe from edge or whatever the parameter is.  
All 'X' spots are the border where the firmware is saying the probe cannot go. You must manually define these locations, are assume them with G29 P3. 

X   X   X   X  X
X   A   A   A  X
X   A   A   A  X
X   A   A   A  X
X   X   X   X   X

If you set the parameter to 0, you likely will have this setup below, like I did. The X is where the probe cannot reach due to the probe offset to nozzle. For example, the upper corner could be 200/200, but for the probe to get there you may have to travel to 240/220, which might be outside your stroke limits. Thus, it cannot probe that location. Once again, fill by G29 P3 or manually entering values (I forget the G29 parameter to use nozzle to finish the mesh).

A   A   A   A  X
A   A   A   A  X
A   A   A   A  X
A   A   A   A  X
A   A  A   A   X

I also had to apply a Y offset for my printer. The stock 0 position was having the nozzle off the bed entirely, so the probe would attempt to hit air and crash the nozzle into the bed. I resolved this by changing the Y_MIN value in the firmware from 0 to '-10'. That way, when the Y axis homes using the switch, it will automatically know it is at the '-10' position, and move 10mm towards the bed, planting the nozzle on the corner.  This will also tell the probe to hit that position when creating the mesh. (Assuming you have the firmware set up to probe the whole bed, and not say just a subset of the bed, such as 100x100 of the 200x200 bed)

stableconnection, 2020-04-29:
running a delta with edge already 0 by know, it probes more now but not even close to the full amount of probes it should. also running bl touch and the offset is correct, still first probe is not at center (supposed to be like this maybe?). or did i miss a point where it says no deltas with UBL?

stableconnection, 2020-04-29:
so at the end of the day everything is set up correctly as far as i can tell and it still only probes 31/64. Did you say i can assume the remaining 33 probes by G29 P3?
and for the not probing the edge issue, i am probing the very edge of the bed no problem, it just stops at a certain point and goes back to center without giving me an error

RFBomb, 2020-04-29:
I don’t see why you can’t run UBL on a delta.

But honestly, I’ve never seen my Cartesian printer start G29 at the center
I don’t think. I believe it always started in the corner when using ‘G29
P1’. But I also don’t have a circular bed like a delta does, so I don’t
know how that would run it.

Are you using the BLTouch to determine Z0 ? I had some trouble with mine
when I was attempting to use Z0 switch and BL Touch separately, so now I
just use the BLTouch for homing.

Also, from the UBL page:
https://marlinfw.org/docs/gcode/G029-ubl.html

   - In most cases the probe can’t reach all areas that the nozzle can due
   to the offsets specified by X_PROBE_OFFSET_FROM_EXTRUDER and
   Y_PROBE_OFFSET_FROM_EXTRUDER. Deltabots can only probe within the area
   where DELTA_PROBEABLE_RADIUS and DELTA_PRINTABLE_RADIUS overlap.

RFBomb, 2020-04-29:
Post your results from “G29 T”

On Wed, Apr 29, 2020 at 12:39 PM robbery525 . <robbery525@gmail.com> wrote:

> I don’t see why you can’t run UBL on a delta.
>
> But honestly, I’ve never seen my Cartesian printer start G29 at the center
> I don’t think. I believe it always started in the corner when using ‘G29 [...]

stableconnection, 2020-04-29:
Bed Topography Report:
    (  -69, 69)                                              ( 69, 69)
        0       1       2       3       4       5       6       7
 7 |   .       .       .       .       .       .       .       .
   |
 6 |   .       .       .       .       .       .       .       .
   |
 5 |   .       .     -0.306  +0.087 [+0.061] +0.046    .       .
   |
 4 |   .     +0.054  +0.087  +0.095  +0.126  +0.117  +0.099    .
   |
 3 |   .     +0.044  +0.076  +0.102  +0.164  +0.174  +0.106    .
   |
 2 |   .     +0.030  +0.071  +0.099  +0.107  +0.148  +0.084    .
   |
 1 |   .     +0.069  +0.109  +0.079  +0.091  +0.180  +0.096    .
   |
 0 |   .       .       .     +0.091  +0.098    .       .       .
        0       1       2       3       4       5       6       7
    (  -69,  -69)                                              ( 69,  -69)

stableconnection, 2020-04-29:
ah, is there an option for a circular bed instead of a square one? if yes where would it be?

stableconnection, 2020-04-29:
problem is probably the FW setting of the bed, it is a circular bed with the size of printable radius but the ubl is trying to probe a square one and thus cant get to the "corners", hence missing an awful lot of probes

RFBomb, 2020-04-29:
That seems to be fairly sound logic.
Mind uploading your config.h file? (You may have to zip it up)

stableconnection, 2020-04-29:
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/4553079/Configuration.h.zip)

stableconnection, 2020-04-29:
if you find major fuckups lemme know, i am a sucker for feedback

stableconnection, 2020-04-29:
> Are you using the BLTouch to determine Z0 ? I had some trouble with mine when I was attempting to use Z0 switch and BL Touch separately, so now I just use the BLTouch for homing.

no problems there

> - In most cases the probe can’t reach all areas that the nozzle can

with my setup the probe should be able to reach every point though

RFBomb, 2020-05-01:
OK, so I have taken a look at your configuration.h file, and have also normalized your topography report to be easier to read. 

`Topography Report
( X -69, Y 69) 													( X 69, Y 69)
	0 	1 		2 		3 		4 		5		6 		7
7 | 	. 	. 		. 		. 		. 		. 		.	 	.
6 | 	. 	. 		. 		. 		. 		. 		. 		.
5 | 	. 	. 		-0.306 	+0.087 	[+0.061] 	+0.046 	. 		.
4 | 	. 	+0.054 	+0.087 	+0.095 	+0.126 	+0.117 	+0.099 	.
3 | 	. 	+0.044 	+0.076 	+0.102 	+0.164 	+0.174 	+0.106 	.
2 | 	. 	+0.030 	+0.071 	+0.099 	+0.107 	+0.148 	+0.084 	.
1 | 	. 	+0.069 	+0.109 	+0.079 	+0.091 	+0.180 	+0.096 	.
0 | 	. 	. 		. 		+0.091 	+0.098 	. 		. 		.
	0 	1 		2 		3 		4 		5 		6 		7
( X -69, Y -69) 												( X 69, Y -69)`

Here are some relevant lines from your config, just to make referencing them in any subsequent comments easier:
`
#define DELTA_PRINTABLE_RADIUS 70.0     // (mm)
--
#define X_BED_SIZE ((DELTA_PRINTABLE_RADIUS) * 2)    // = 140
#define Y_BED_SIZE ((DELTA_PRINTABLE_RADIUS) * 2)    // = 140
--
#define X_MIN_POS -(DELTA_PRINTABLE_RADIUS)     // = -70
#define Y_MIN_POS -(DELTA_PRINTABLE_RADIUS)    // = -70   
#define X_MAX_POS DELTA_PRINTABLE_RADIUS       // = 70
#define Y_MAX_POS DELTA_PRINTABLE_RADIUS      // = 70
--
#define NOZZLE_TO_PROBE_OFFSET { 0.0, 25.0, -1.70 }
`
[UBL Stuff]
`
 #define MESH_INSET 1              // Set Mesh bounds as an inset region of the bed
#define GRID_MAX_POINTS_X 8      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
`

-------------------------------------------------------------------------------------------------
`Delta_Probable_Radius` appears to have been removed (replaced by the `MESH_INSET` value.) So my earlier comment can be disregarded. Your topography report appears to jive with your `MESH_INSET` and `DELTA_PRINTABLE_RADIUS ` settings, so I'm not concerned with that. 

All of the `.` locations in the bed topography report are undetermined heights. The corners won't be able to be determined due to a square grid with circular bed. You can assume these values with `G29 P3` to fill them in, or you can just use `G29 A` in place of `M420 S1`. 

As for it not being a complete circle, this is due in part from your `NOZZLE_TO_PROBE_OFFSET ` values. You can see that 1-6 columns (X) have values, and 0-5 rows (Y) have values. The missing rows would be due to the Y offset of 25. The missing columns are likely due to the printer thinking it cannot reach that location with the probe (due to the Y offset). For example, your firmware specifies the  `Y_MAX_POS DELTA_PRINTABLE_RADIUS`, or ` Y_MAX_POS 70` if you were to put the number in. Your Y offset is 25. So in order to probe any nozzle locations at Y70, your print head would actually need to be at Y95, which is out of bounds according to the ` Y_MAX_POS` setting of 70. This is something I also fought/ was confused by when I intially installed my BLTouch, prior to learning about the limitations here. 

-------------------------------------------------------------------------------------------------
What I would try first is using `G29 P1 C`. This will have the printer attempt to probe any unknown points it thinks it can get to. Follow that with `G29 P2` (Required LCD Controller!). P2 will allow the printer to move the nozzle to an unknown grid point, and you can lower the nozzle to the desired height for the mesh.  If you think the mesh itself is fine, you can use `G29 P3` to assume the values of unknown points (based on known points). 

To reiterate and answer your original questions:
>      if no, any ideas why it would stop at ~half?

UBL appears to be working properly in your case. Any locations unknown are likely unable to be reached with the probe. See explanation paragraph above. 

>     why is the head moving in the wrong direction with G29 to get to center?
Still not sure what you mean by 'wrong direction'. To probe approximately ( 0 , 0 ), your actual position would be about ( 0 , 25 ), based on the offsets you provided.  

>     Also, what could be the reason for my setup not to take the G29 standalone command but only in combination with e.g. P1

This is how the command works. 
https://marlinfw.org/docs/gcode/G029-ubl.html

> first probe is not at center 
UBL takes the amount of grid points you assign it, divides the bed up into a grid, and works from there. It tries to cover the grid the best it can, 0,0 may not be hit, but if ( 3,3) (-3 , -3) (-3 , 3) (3 , -3) are all hit (for example) then it will give a really good approximation of (0,0)

-------------------------------------------------------------------------------------------------
Sidenote: 
You have `#define Z_SAFE_HOMING` enabled, along with `#define Z_HOME_DIR 1 //Home in MAX direction` .  Since you are homing a delta machine (and moving to the positive stops), `Z_SAFE_HOMING` isn't doing anything for you at all. You can disable that, as it isn't ever used anyway. (Took a look at the Delta homing routine to double check. Z_Safe_Homing is only used for cartesians, and then only if homing in negative direction. 

 You also have `#define DEBUG_LEVELING_FEATURE` enabled, which you likely won't need once all is working properly.