MarlinFirmware/Marlin #17103: [BUG] G26 prints outside of bed

tbe, 2020-03-09:
### Bug Description

During the mesh validation print, G26 prints outside bed, if X_MAX or Y_MAX are outside of the bed.

### My Configurations
```
// The size of the print bed
#define X_BED_SIZE 235
#define Y_BED_SIZE 235

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 245
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 250

```

### Steps to Reproduce

1. Define a MAX_POS outside of the bed size
2. Do a G26 validation print

**Expected behavior:** marlin respects `X_BED_SIZE` and `Y_BED_SIZE` 

**Actual behavior:** Marlin ignores the bed size and only uses the max pos

#### Additional Information

I use a BLTouch on the left side of my extruder on my ender 3 pro. As the X rail is long enought to move the hotend outside of the bed, so the probe can probe more of the actual bed, i have redefined `X_MAX_POS`

thinkyhead, 2020-03-11:
According to my reading of the code the XY points in the `G26` mesh are supposed to be derived from the mesh coordinates. The same coordinates used by `G29` for probing.

Please supply your configurations, preferably as a single ZIP file, so that we can see what's up with your mesh boundary.

tbe, 2020-03-11:
I attached my config. 

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4318684/config.zip)

thinkyhead, 2020-03-12:
See if this removing this line from `gcode/bedlevel/G26.cpp` helps…
```diff
  if (current_position.z < Z_CLEARANCE_BETWEEN_PROBES) {
    do_blocking_move_to_z(Z_CLEARANCE_BETWEEN_PROBES);
-   current_position = destination;
  }
```

tbe, 2020-03-13:
Still prints outside of the bed. 
[image]
[image]

I had a look at the code, and as far as i could see, it shouldn't be possible that a mesh point is calculated outside the printable area. :confused: 

What i have seen is, that `position_is_reachable` is used to check if a point is printable. But this function returns just if the head can be moved to this position, not if it is still in the bed boundaries.

ellensp, 2020-03-13:
Verify location of X0,Y0, is this the very corners of your bed?
please lay a ruler over your bed in X direction. verify it is at least 235 from X0 position.

tbe, 2020-03-13:
> Verify location of X0,Y0, is this the very corners of your bed?
Yes. As seen in the first picture, the mesh starts at at 1,1.
> please lay a ruler over your bed in X direction. verify it is at least 235 from X0 position.
The glass plate used is a little big larger, around 236.5mm each side.

thinkyhead, 2020-03-14:
Your bed might be 236.5mm but it looks like when X=0 the nozzle is about 2-3mm from the left edge of the bed. So you really only have about 230mm to 233mm of actual usable space.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle