MarlinFirmware/Marlin #17104: [BUG] SD card support stopped working in 2.0.4.4, BTT SKR 1.3

brettvitaz, 2020-03-10:
### Bug Description

After upgrading from 2.0.4.3 to 2.0.4.4 on my BTT SKR 1.3, SD card support stopped working.

### My Configurations

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4309821/configuration.zip)

### Steps to Reproduce

1. Configure the following values:
* #define SDSUPPORT
* #define SDCARD_CONNECTION ONBOARD
2. Compile, load new firmware

**Expected behavior:** 
Open menu and print from SD should work.

**Actual behavior:** 
Open menu and get message "Media Released". Clicking "Attach Media" does not work.

brettvitaz, 2020-03-10:
SD support works properly when copying configuration to a clean 2.0.4.3.

Powerprobot, 2020-03-10:
the change in [#17054](https://github.com/MarlinFirmware/Marlin/pull/17054) make the SD unuseable.

The changed file is cardreader.h

This doesn't work for me. I have the SRK 1.3 and ReprapFullGraphicsDisplay with Card-Reader. The old version works fine. With the new line toggle the SD Status in wrong way; if i put the SD in, are displayed "Medium removed" and can't print from SD(LCD). I comment out the new Line and add the old ones; this works.
Update:
I change the Logic of the Line in '==':

#define IS_SD_INSERTED() (READ(SD_DETECT_PIN) == ENABLED(SD_DETECT_INVERTED))

This works fine.

thinkyhead, 2020-03-11:
Patch applied. Expect 2.0.5 to be out soon.

closed 2020-03-11 by thinkyhead