MarlinFirmware/Marlin #17106: [BUG] Click to resume... on sd print Marlin 2.0.4.4 bugfix

Hagbard-84, 2020-03-10:
### Bug Description 
At the end of a SD print i get Click to resume... on the lcd screen

### My Configurations

Marlin 2.0.4.4 bugfix.   

//#define RGB_LED
//#define RGBW_LED is disabled

and ware did the JERK  settings go? only showing one setting

problem not preset on 2.0.3

Hagbard-84, 2020-03-10:
#

DJDOUBLEG, 2020-03-10:
Same error to me.
SKR pro v1.1 + Marlin 2.0.4.4
and 
SKR v1.3 + Marlin 2.0.4.4
[image]
[image]

ellensp, 2020-03-10:
See https://github.com/MarlinFirmware/Marlin/issues/16808
Is fixed in bugfix, just missed the 2.0.4.4 release.

thinkyhead, 2020-03-11:
Other end-of-SD issues are patched in bugfix as well. Expect 2.0.5 to be out soon.

closed 2020-03-11 by thinkyhead

bigtreetech, 2020-03-12:
https://github.com/MarlinFirmware/Marlin/blob/1f55986ed2aeb9db30b2438b125aa8ce64a5f09f/Marlin/src/MarlinCore.cpp#L441
In fact, the latest bugfix also has this issue. after printing, it will execute "M31", "M77" and then "M0", and then it will pause