MarlinFirmware/Marlin #17110: Marlin does not send "File opened" message to host (Octoprint)

Miraculix200, 2020-03-10:
I'm not sure if this is a problem with Octoprint or Marlin 2.x, but the problem didn't occur with the original firmware of my Sapphire Pro, or with Marlin 1.x on my other printer.

The problem is that the printer never signalizes to the host when I start printing from SD card on the touch display of the printer. It does send the "echo:enqueueing "M23 xxxx~1.gco" and M24 message, but not the "File opened" message.

The host does receive other messages, e.g. temperatures and "SD printing byte" messages. 

Communication works properly when starting a print through Octoprint, but I prefer starting prints from the display of the printer.

The bug(?) occured with the Marlin 2.x bugfix release and still occurs with Marlin 2.0.4.4.

thinkyhead, 2020-03-11:
Are you sending the file open command from OctoPrint or from elsewhere? Marlin sends serial output to the default serial connection. If there are two serial ports, Marlin will send the "File opened" response to the serial device that initiated the SD print.

```cpp
  if (file.open(curDir, fname, O_READ)) {
    filesize = file.fileSize();
    sdpos = 0;
    SERIAL_ECHOLNPAIR(STR_SD_FILE_OPENED, fname, STR_SD_SIZE, filesize);
    SERIAL_ECHOLNPGM(STR_SD_FILE_SELECTED);

    selectFileByName(fname);
    ui.set_status(longFilename[0] ? longFilename : fname);
  }
  else
    openFailed(fname);
```

thinkyhead, 2020-03-11:
OK. I've patched Marlin to make SD card file open into a broadcast instead of just a reply.

closed 2020-03-11 by thinkyhead

Miraculix200, 2020-03-11:
Thank you. For anyone who is interested how to possibly solve the problem without patching: 

In my Configuration.h these lines 
#define SERIAL_PORT 3
#define SERIAL_PORT_2 1

have been changed into 

#define SERIAL_PORT 1
#define SERIAL_PORT_2 3

and now it works as supposed to.