MarlinFirmware/Marlin #17111: [BUG] Azteeg X5 GT: BSD2660 drivers will not work when LCD (miniViki + wifi) is connected

xnopasaranx, 2020-03-10:
### Bug Description

It seems that there is still a problem with SPI bus management. 
The azteeg x5 gt is supposed to work fine with the miniViki display, but I cannot get the stepper drivers to work when the LCD is connected. Also when both are connected to the board, I can see noise on the display when I power up, before the Marlin Boot Logo is shown.
Marlin shows "TMC connection error" on screen and sending M122 reports: 
```
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW
```
The drivers are fine and respond when the display is not connected.

I own the latest miniViki version which comes with a builtin wifi module (based on ESP3D), not sure if that revision has changed anything SPI related

I would like to try TMC_USE_SW_SPI but there are no defaults for the pins in Marlin\src\pins\lpc1769\pins_AZTEEG_X5_GT.h so I have no idea what I should define these as.

### My Configurations
[Marlin-conf.zip](https://github.com/MarlinFirmware/Marlin/files/4314234/Marlin-conf.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Compile Marlin with miniViki LCD enabled and install BSD2660 steppers on the board
2. Upload firmware and power on
3. Watch the LCD come up and show "TMC connection error"

**Expected behavior:** LCD works fine and stepper drivers report: OK

**Actual behavior:** LCD comes up with some noise on screen, then shows marlin logo and seems fine, stepper drivers do not respond and M122 shows that they are unreachable

#### Additional Information
Here is a gist with the result of M43 with and without LCD enabled before building:
[https://gist.github.com/xnopasaranx/86a1a606bc6db26a42960f81109b2bb7](url)

Not sure if this is helpful, but more info is always better than less.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-03-14:
I noticed today that a lot of lpc1769 boards have "#define FORCE_SOFT_SPI" in their pins files, but not the Azteeg X5 GT.  Give this a try.

xnopasaranx, 2020-03-14:
@ellensp Hi thanks for the advice. Sadly that did not help and I still get the "TMC connection error".

xnopasaranx, 2020-05-17:
Pulling the latest bugfix-2.0.x changes has fixed this problem for me! I had assembled the electronics for the printer completely and tried smoothieware, then compiled marlin 2.0 stable and it was printing eventhough the TMC Connection Error was still thrown, but this feels much safer and the constant error message was bugging me.

closed 2020-05-17 by xnopasaranx