MarlinFirmware/Marlin #17113: [BUG] SKR 1.4 turbo BTT with TMC2209 BTT V1.2 E0 direction

onebehler, 2020-03-11:
### Bug Description

I am having a problem with two seperate BTT SKR 1.4 Turbo paired with BTT TMC2209 V1.2 drivers moving the extruder in both directions. I can send the g1 e10 and g1 e-10 and the motor spins in the same direction.

I have removed the cold extruder block from my firmware, tried several different motors with different cabling, swapped drivers from one that works in both directions. I set the lone jumper per driver slot for UART mode as described in the manual. I am using sensorless homing and bltouch. I have looked at several different web sources to find where my error is. I cloned the github last week and am working with VS and PIO to compile and make. Any help is greatly appreciated.

### My Configurations

Cartesian style printer with dual z axis.  Converted from non-marlin controller to SKR1.4 turbo with TMC2209 drivers.  All from BTT.  Using VS and PIO to configure and load Marlin.  Started with cloned github from BTT in the master branch for SKR 1.3/1.4  Configured for sensorless homing and BLTouch.  Steps not currently accurate in config files.

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step] Tried inverting steps and disabling nozzle low temp block
2. [Second Step] Swapped motors, leads, and drivers
3. [and so on...] Moved all drivers to new board with only E0 connected.  Motor works as expected in all location except E positions.  Did not test E1

**Expected behavior:** I am expecting to be able to extrude and retract on the E0

**Actual behavior:** E0 only moves in one direction with both positive and negative gcode or buttons via OctoPi

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4315729/Config.zip)

ellensp, 2020-03-11:
"Started with cloned github from BTT" Then you would normally need to get help from BTT. 
But In this case I case I know what is going on. 
If you had you started from real marlin from this github it would have told you what is going on.

You have #define SERIAL_PORT 2
These are the serial ports on your controller
 ```
*   Port  |  TX   |  RX   |
 *    ---  |  ---  |  ---  |
 * Serial  | P0_02 | P0_03 |
 * Serial1 | P0_15 | P0_16 |
 * Serial2 | P0_10 | P0_11 |
 * Serial3 | P0_00 | P0_01 |
```
E0 Direction line is P0_11, ie it conflicts with Serial2

onebehler, 2020-03-11:
Ok, I have so many different questions.  Do I need to define serial 2 then?  I am using OctoPi on the USB.  If I comment out #define SERIAL_PORT 2, will that correct the issue? Thank you for your help on this.  I really appreciate that.

onebehler, 2020-03-11:
That was the problem.  I changed #define SERIAL_PORT  back to 0 from 2 and now I have bi-directional control.  Thank you so very much for your help on this.  I will mark closed.

closed 2020-03-11 by onebehler