MarlinFirmware/Marlin #17116: [BUG] Z runs away during home if motion command during homing

AnimeEd, 2020-03-11:
### Bug Description

When during the homing process, if you go to motion and give a motion command, the Z will run away during the second slow homing action.

### My Configurations

Ender 3 Pro, Marlin 2.0.4.4

### Steps to Reproduce

1. Move print ahead away from home do that it will take some time to home
2. Tell printer to find home using knob
3. Before homing is finish, go to motion > X/Y/Z > 10mm/1mm/0.1/0.025mm > turn knob to issue command
4. When printer get to Z homing, it will go down quickly to find the end switch, it will then go up and typically come back down slowly to precisionly find the end switch. Instead of going up and then down slowly, it will continue going up unless you kill power

**Expected behavior:** [What you expect to happen]
Printer should home correctly and then run motion commands as issued

**Actual behavior:** [What actually happens]
Printer will run away during homing

#### Additional Information
Can provide video of bug if requested

AnimeEd, 2020-03-14:
it appears this also applies to motion commands while homing to the extruder. The Z axis also takes up the rate of the axis you gave command to. If i ask for 10mm in X or Y while homing, the Z will run away at a fast rate. If I ask for 10mm in extrusion while homing, Z will run away much slower.

boelle, 2020-06-20:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

AnimeEd, 2020-06-21:
I'm on bugfix-2.0.x June 10 2020 now and it still exist
What appears to be happening is that the "Motion" commands are being injected into the Auto Home algorithm before it finishes.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4810356/Marlin.zip)

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle