MarlinFirmware/Marlin #17117: [BUG] (short stop  during printing - zits & blobs)

DanielSan1000, 2020-03-11:
### Bug Description

During the printing extruder stoping move on short moment and this causes blobs and zits.
I used gcode where on prevous marlin version was ok. Its look that processor isn't able to caclucalte on right time some parts of the g-code.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/4316597/conf.zip)

### Steps to Reproduce

**Expected behavior:**
It should work smoothly without any break and stops

**Actual behavior:** 
Surface have visbile  blobs and zits isnt like was before

**3D printer:** 

Ender 3 & SKR E3 mini v1.2, Marlin 2.0.4.4

superperf, 2020-03-12:
i watched a similar behavior on my ender 5 with skr1.4 and Marlin 2.0.3.

I watched on the Console and it seems that after every G92 E0 the Printer stops for one small moment and leaves a zit on the outer perimeter. I also watched some stuttering when travelling longer ways.

Got the Marlin bugfix branch before with an SKR1.3 and didn't had this Issues above on this Board/Marlin Version

DanielSan1000, 2020-03-13:
i found some move which touch these problem wasnt solve it ist beter

DanielSan1000, 2020-03-13:
I found some solution but for my case isn't work perfectly its better but .. Currently  I'm playing with paramters of the Mesh Resolution.
CNC Kitchen - > https://youtu.be/Hvw3DrVAeTA?t=2

superperf, 2020-03-13:
thanks. going to look forward to solve the issue. hopefully the nightly builds have fixed some of this. thanks for the link

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle