MarlinFirmware/Marlin #17118: [BUG] due and serial connection

maukcc, 2020-03-11:
### Bug Description

serial connection over octoprint to bugfix2 is OK. when disconnecting and re-connecting, there is no more connection possible.

### My Configurations

bugfix2(on due) connected to octoprint over SERIAL_PORT 2

### Steps to Reproduce

connect to marlin
manually disconnect
re-connect to marlin

**Expected behavior:** [What you expect to happen]
re-connection will happen

**Actual behavior:** [What actually happens]
re-connection does not happen

#### Additional Information
re-connection only possible after the reset button on the due is pressed.

thinkyhead, 2020-03-11:
I've noticed that some boards (LPC176x) don't care when you connect or disconnect. They just keep the connection alive as if you never connected or disconnected.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle